qmk/qmk_firmware #5072: Traduce keymap into specific matrix

lebarde, 2019-02-07:
Hello,
In order to use my 18-pin Pro Micro board on a 15×5 keyboard project, I have designed a [quite complex matrix that uses 11 rows and 7 cols](https://deskthority.net/viewtopic.php?p=436812#p436812).

As I need to write several layers, I would like a facility tool to enhance the keymap readability (*and writing*) inside the `keymap.c` file. I imagine defining a keymap matrix as it is done yet, then passing it into a function that traduces that matrix (here, 15×5) into a matrix that will fit the actual wiring.

I think this is feasible with a simple C function, but I don't know how to proceed. What do you think of it?

Thanks!!

pelrun, 2019-02-07:
That's literally what the LAYOUT macros do, convert a keyboard-shaped keymap into a matrix shaped one.

lebarde, 2019-02-07:
Thank you @pelrun for your answer. I'll get into this!

drashna, 2019-02-08:
Specifically, you want this page: https://docs.qmk.fm/#/hand_wire?id=ltproject_namegth

That goes over how it's laid out. 

Also, @mechmerlin has some great videos of him setting up a number of keyboards, including how to set up the layout macro for the keyboard. 
https://www.youtube.com/channel/UCdfrYMwAJ8LHvy8-j_WIxAw/videos

Also, the planck rev6 is a good example of this.

lebarde, 2019-05-12:
I had found a way to get into it, and finally my problem was solved. Thank you!

closed 2019-05-12 by lebarde