qmk/qmk_firmware #5083: make git-submodule fails in

atrnh, 2019-02-08:
**Describe the bug**

**System Information**
 - Keyboard:  
   - Revision (if applicable):  
 - Operating System:
 - avr-gcc version: 

 - arm gcc version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] Auto Hot Key
   - [ ] Karabiner
   - [ ] Other

**Additional context**

zvecr, 2019-02-08:
Version of git on ubuntu 16.04 is old, #4905 seems to fix this as debian 9 ships a newer git (version 2.11).

atrnh, 2019-02-09:
Removing `--progress` from `make git-submodule` seems like a better idea.

Switching from Ubuntu to Debian just to use `--progress` is overkill.

zvecr, 2019-02-09:
I agree with both statements, upgrading base image and the git dependency within make. My comment was more to link the two issues as i had also observed the same problem, and note the potential workaround.

From the man page:
```
--progress
Progress status is reported on the standard error stream by default when it is attached to a terminal, unless -q is specified. This flag forces progress status even if the standard error stream is not directed to a terminal.
```
Not sure if removing the flag will have any unintended side effects.

atrnh, 2019-02-09:
@zvecr Right on, thanks for the clarifcation!

atrnh, 2019-02-09:
I assume the lack of verbosity from leaving out `--progress` can make it seem like git is hanging and not doing anything. Only solution I can think of is `echo`-ing some reassuring messages 🤔

zvecr, 2019-10-28:
Fixed by #4905.

closed 2019-10-28 by zvecr