qmk/qmk_firmware #5118: KBD75 Rev2 board - Caps lock LED code not working?

necropsyuk, 2019-02-13:
I have used both Rev1 and Rev2 builds on my KBD75 rev2 (USB-C) board. The caps lock LED is not functional. 

This is the code that appears in both the rev1.c and rev2.c files

```
void led_set_kb(uint8_t usb_led) {
	// put your keyboard LED indicator (ex: Caps Lock LED) toggling code here
  if (IS_LED_ON(usb_led, USB_LED_CAPS_LOCK)) {
    writePinHigh(B2);
  } else {
    writePinLow(B2);
  }

	led_set_user(usb_led);
}
```

I know the LED itself is functional as whatever build was on the PCB out the box from kbdfans was working well.

Is it possible that the pin is different? If so then how do I figure out which pin code relates to which key?

fauxpark, 2019-02-14:
According to [this drawing](https://cdn.shopify.com/s/files/1/1473/3902/files/Q75.pdf?9073386096780420683) B2 is correct (it's the second pin to the right of the "U1" text).

The circuit:
[image]

It seems like the pin should actually be pulled *low* to turn the LED on. But you should be seeing the LED light up when Caps Lock is off instead of "not functional".

necropsyuk, 2019-02-14:
> According to [this drawing](https://cdn.shopify.com/s/files/1/1473/3902/files/Q75.pdf?9073386096780420683) B2 is correct (it's the second pin to the right of the "U1" text).
>
> The circuit:
> [image] [...]

Unfortunately it's off at all times. I did try swapping the commands in the if statement to pull to low, else high. Still nothing. 

Interestingly, when I comment out this code, the LED still does not light, even though other LEDs do. I'm at a loss.

drashna, 2019-02-14:
How is the pin being set? 

Try adding the following to the rev2.c file: 
```c
void matrix_init_kb(void) {
  setPinOutput(B2);
  matrix_init_user();
}
```

necropsyuk, 2019-02-14:
> How is the pin being set?
> 
> Try adding the following to the rev2.c file:
> 
> ```c
> void matrix_init_kb(void) {
>   setPinOutput(B2);
>   matrix_init_user();
> }
> ```

Yes! It worked. Legend. Thanks @drashna 

Final working code in rev2.c for anyone else as follows:

```
#include "rev2.h"

void matrix_init_kb(void) {
  setPinOutput(B2);
  matrix_init_user();
}

void led_set_kb(uint8_t usb_led) {
	// put your keyboard LED indicator (ex: Caps Lock LED) toggling code here
  if (IS_LED_ON(usb_led, USB_LED_CAPS_LOCK)) {
    writePinLow(B2);
  } else {
    writePinHigh(B2);
  }

	led_set_user(usb_led);
}
```

closed 2019-02-14 by necropsyuk

drashna, 2019-02-14:
Awesome! 

Well, if you want, open a PR that fixes this (ideally, in its own branch).  For both the rev1 and rev2, since both seem to have this issue. 

If not, I'm sure that myself or somebody else can.

necropsyuk, 2019-02-14:
> Awesome!
> 
> Well, if you want, open a PR that fixes this (ideally, in its own branch). For both the rev1 and rev2, since both seem to have this issue.
> 
> If not, I'm sure that myself or somebody else can.

Thanks for doing it, this is my first time using Github. Nice to know my issue might help others though.

CUTTY22, 2020-01-13:
Hi i have a question. i am new and i am having the same problem with my caps lock led not working on my kbd67 rev2 once i flashed it using qmk. how do i access or where to i go to find the rev2.c file so i can copy and paste that code up above to get the caps lock led to work. thank you very much.

necropsyuk, 2020-01-13:
> Hi i have a question. i am new and i am having the same problem with my caps lock led not working on my kbd67 rev2 once i flashed it using qmk. how do i access or where to i go to find the rev2.c file so i can copy and paste that code up above to get the caps lock led to work. thank you very much.

https://github.com/qmk/qmk_firmware/tree/master/keyboards/kbdfans/kbd67/rev2

Here you go. Not sure if it's been updated, but you can amend it as per my above changes.

CUTTY22, 2020-01-13:
thank you. sorry about another question. i found the code line but do i have to install like a build tool or something to amend it. and do i flash it to the keyboard. sorry again this is my first time dealing with this. 

 
 

https://github.com/qmk/qmk_firmware/tree/master/keyboards/kbdfans/kbd67/rev2

necropsyuk, 2020-01-14:
Yes you need to build it. Check out the documentation.

https://docs.qmk.fm/#/

CUTTY22, 2020-01-14:
ok thank you once again ill give it a look. have a great day

 
 

https://docs.qmk.fm/#/

PR #5129 (2019-02-14): Add pin initialization for KBD75's LED

## Description
So that the LED indicator actually works

Specifically, the keyboard sets the pin high/low to set the LED, but never actually sets the output mode, so it doesn't work. 

## Types of changes

- [ ] Core
- [ ] Bugfix
- [ ] New Feature
- [ ] Enhancement/Optimization
- [X] Keyboard (addition or update)
- [ ] Keymap/Layout/Userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by this PR

* Fixes #5118

## Checklist:

- [X] My code follows the code style of this project.
- [ ] My change requires a change to the documentat [... cut]