qmk/qmk_firmware #5131: QMP Toolbox - Ever since the site update, only default keymap for dz60 is available.

spakecdk, 2019-02-14:
Basically the title / When I try to load a custom keymap.hex it ignores it and loads the default. Also, when the dy60 is selected in the drop-down menu, the keymap drop-down menu is greyed-out (disabled). It worked fine on 2nd of February, unless I am missing something,

drashna, 2019-02-14:
For the QMK Toolbox?  As far as I'm aware, it only ever showed the default keymap, for each and every keyboard. 

In fact, looking at the code, that's all it shows (the default)

spakecdk, 2019-02-14:
I see, I just went to check as well and you are right. I now have realised the issue has been that I have misremembered a step about flashing a keyboard (I was pressing the Load button lol, when all I had to do was Flash). Thanks!

closed 2019-02-14 by spakecdk