qmk/qmk_firmware #5139: Modifying ascii_to_keycode_lut for non US-ASCII keymaps

rys, 2019-02-15:
Is it possible to change the `ascii_to_keycode_lut` in user keymaps, to better support non US ASCII?

## Describe the Bug

When using `SEND_STRING()`, which ends up in `send_char()`, there's a lookup into the ASCII keycode LUT to figure out what keycode to send, and then a lookup into the shift LUT to see if `KC_LSFT` should be sent around the target keycode.

The ASCII LUT assumes a US keymap when decoding the string, which isn't always what's being used on the host the board is connected to. For example `SEND_STRING("@");` will generate a `"` on my system since it decodes to:

```
key_down left_shift
key_down 2
key_up 2
key_up left_shift
```
Can user keymaps therefore redefine the LUT?

## System Information

 - Keyboard:  
   - Revision (if applicable): all
 - Operating system: macOS 10.14.2
 - AVR GCC version: 7.3.0

 - ARM GCC version: 7.3.1 20180622

 - QMK Firmware version: 0.6.275-2-g918a52

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [x] Karabiner (but only to check scancodes)
   - [ ] Other:

## Additional Context

drashna, 2019-02-15:
You mean like this? 
https://github.com/qmk/qmk_firmware/blob/master/quantum/keymap_extras/sendstring_german.h

rys, 2019-02-15:
Ha, exactly like that! I’ll create a PR in a similar style for the map I need.

closed 2019-02-15 by rys

drashna, 2019-02-15:
Welcome! :)