qmk/qmk_firmware #5155: Idobo / XD75 reset help

olivierpons, 2019-02-17:
After successful "custom" GMK flash,  the keymap isn't the same, it's the "default" one.
So I guess I should press spacebar and "B" again to reset the keyboard, like I did the first time.
The guy who made the default idobo keymap here "./keyboards/idobo/keymaps/default/keymap.c"
did this on the last row:

KC_LCTL, KC_LGUI, KC_LALT, KC_SPC, KC_SPC, KC_SPC, KC_SPC, KC_SPC, KC_SPC, KC_RALT, KC_RCTL, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT

Yes you've read it: 6 times the space bar.

Now I can't reset my keyboard, even when trying to press the 6 "KC_SPC" keys at the same time + the "b".

Any idea on how to reset my keyboard?

olivierpons, 2019-02-17:
I've just unmounted my keyboard (I bought it 100% mounted, and there's no hole for the reset button...) and I've seen the reset button.

closed 2019-02-17 by olivierpons

nostdm, 2020-10-18:
@olivierpons Did you ever manage to find a solution? I just ran into the same issue - entered the reset mode using space+b and flashed idobo layout with custom keymap and now I can't reset the keyboard the same way as before. If I press the same sequence space+b and plug in the keyboard nothing happens. I also tried other suggested methods on this page https://docs.qmk.fm/#/newbs_flashing and nothing worked (except for the 1st sequence as I haven't mapped a Pause key).

nostdm, 2020-10-18:
Turns out I had to add a Reset button to my custom key map in order to be able to press it as a key sequence. 
Ultimately I had to open the keyboard and reset it using the reset button on the PCB.