qmk/qmk_firmware #5207: mcufont.exe being picked up by AntiVirus

dernPerkins, 2019-02-21:
Investigate into reasons behind why the mcufontencoder under lib\ugfx\tools\mcufontencoder\binaries\windows\mcufont.exe keeps showing up as a threat and being quarantined by malware protection.

Symantec reports it as a Heur.AdvML.C threat: is a heuristic detection designed to generically detect malicious files using advanced machine learning technology. A file detected by this detection name is deemed by Symantec to pose a risk to users and is therefore blocked from accessing the computer. 

As well as PUA.Gen.2: is a generic detection for many individual but varied potentially unwanted applications for which specific definitions have not been created. A generic detection is used because it protects against many potentially unwanted applications that share similar characteristics.

neurot1cal, 2019-03-30:
Same thing for Mcafee...

[image]

kainoaseto, 2019-07-24:
Sophos flagged this as well, I tend to error on the side of better safe than sorry but I imagine this would cause issues for some of the windows users. Maybe a rollback of a previous version of the ugfx would work until we find out what's going on

noroadsleft, 2019-07-24:
Actually, it's a goal of QMK to remove uGFX as a dependency, due to licensing issues.

No work has been done yet on this front, but it's a definite desire.

kbelcher83, 2019-11-12:
Has there been any progress on this? Many of the major AV players are picking it up https://www.virustotal.com/gui/file/727952580c14ecb0f39408aa3ffe568b5486c55dd5b91f9e86a167263e289a8b/detection

If it gets quarantined, can qmk still compile keyboards?

noroadsleft, 2019-11-12:
@kbelcher83 For progress, not that I know.

The only keyboards in the repository that are affected by this are the Ergodox Infinity and the WhiteFox – none of our other supported boards use uGFX at all. QMK would still be able to compile any other boards, though.

jandusek, 2020-06-18:
Any news on this front? If that dependency isn't needed, can it be dropped?

jwithington, 2021-05-25:
Still being flagged by Sophos

noroadsleft, 2021-08-09:
I believe @fauxpark is working on removing the uGFX dependency, but I don't recall the status of that effort.

fauxpark, 2021-08-09:
firetech is doing most of the work as they actually have an Ergodox Infinity to test on, I think at this point we are just waiting for the develop merge, then we can continue with the next phase which will probably end with the submodule being removed.

fauxpark, 2022-04-07:
uGFX has been removed now, so this issue should be resolved.

closed 2022-04-07 by fauxpark