qmk/qmk_firmware #5224: error:

fruitloop, 2019-02-23:


fruitloop, 2019-02-23:
Hi, I tried to use the Leader Key functionality, but I can't get it to work

I set my layout like this:
```
  [_BASE] = LAYOUT(
    TAB_FN1, KC_Q,    KC_W,    KC_E,    KC_R,    KC_T,    KC_Y,    KC_U,    KC_I,    KC_O,    KC_P,    KC_LBRC, KC_RBRC,
    CTL_ESC, KC_A,    KC_S,    KC_D,    KC_F,    KC_G,    KC_H,    KC_J,    KC_K,    KC_L,    KC_SCLN, ENT_FN9,
    KC_LSPO, KC_Z,    KC_X,    KC_C,    KC_V,    KC_B,    KC_N,    KC_M,    KC_COMM, KC_DOT,  KC_SLSH, KC_RSPC,
    //FN2,     KC_LALT, KC_LGUI, KC_NO,   SP_FN3,  BS_FN4,  FN5,     FN6,     FN7,     FN8
    FN2,     KC_LALT, KC_LGUI, KC_NO,   SP_FN3,  BS_FN4,  FN5,     FN6,     KC_LEAD,     FN8
  ),

```

but I get this error:
```
Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  16376	      0	  16376	   3ff8	.build/keebio_laplace_roland.hex

Compiling: keyboards/keebio/laplace/keymaps/roland/keymap.c                                        In file included from keyboards/keebio/laplace/keymaps/roland/keymap.c:1:
keyboards/keebio/laplace/keymaps/roland/keymap.c:63:77: error: 'KC_LEAD' undeclared here (not in a function); did you mean 'KC_CLEAR'?
     FN2,     KC_LALT, KC_LGUI, KC_NO,   SP_FN3,  BS_FN4,  FN5,     FN6,     KC_LEAD,     FN8
                                                                             ^~~~~~~
keyboards/keebio/laplace/laplace.h:18:22: note: in definition of macro 'LAYOUT'
         {KC_NO, D13, D12, D11, D10, KC_NO, KC_NO} \
                      ^~~
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [tmk_core/rules.mk:372: .build/obj_keebio_laplace_roland/keyboards/keebio/laplace/keymaps/roland/keymap.o] Error 1
Make finished with errors

```

fauxpark, 2019-02-23:
You probably need to set `LEADER_ENABLE = yes` in your rules.mk. `KC_LEAD` is only declared as a keycode when the feature is enabled.

fruitloop, 2019-02-24:
Thanks a lot, that was the case, this option was not in the rules.mk for my keyboard (laplace 40) and there was no mention of that option in the corresponding feature documentation. (at least on https://beta.docs.qmk.fm/features/feature_leader_key this is fixed now)

closed 2019-02-24 by fruitloop