qmk/qmk_firmware #5262: Media keys don't work in ACTION_TAP_DANCE_DOUBLE on preonic/rev3

evantravers, 2019-02-26:
This keymap works on my planck/light and lets_split_eh/eh.

https://github.com/evantravers/qmk_firmware/commit/e3d19273730851f68c6f5dde106758016c609a8b

On my preonic/rev3, this is what happens at the moment:

```
ACTION_TAP_DANCE_DOUBLE(KC_LCTL, KC_MNXT) # both work
ACTION_TAP_DANCE_DOUBLE(KC_MPRV, KC_MNXT) # only the double tap works (next), single press fails
ACTION_TAP_DANCE_DOUBLE(KC_MNXT, KC_MPRV) # now prev is doubletap and it works. Next doesn’t fire. 
```

I'm a little at a loss.

evantravers, 2019-03-04:
So… I just realized that there are separate media keys for OSX and windows… I'm going to try the OSX ones here and see if it makes a difference.

It doesn't explain why the current configuration works with a Let's Split Eh and Planck/Light… but it's worth a shot.

evantravers, 2019-03-04:
Well… it tried it… https://github.com/evantravers/qmk_firmware/commit/37c4a5f4e5f94e82948f796e0067e65172a5a95c

It still doesn't work on the preonic/rev3. I am at a loss. :(

drashna, 2019-03-18:
Well, I know that Tap Dance doesn't work right on ARM boards, ATM.  That's most likely what you're seeing here. Unfortunately.

evantravers, 2019-05-23:
@drashna :(

Has there been any progress made on that that I can code review or help test?

drashna, 2019-05-23:
yes and no.   To fix the issues with ARM, I think it requires updating ChibiOS. Which is NOT a small feat.   

There are a couple of users that have talked about it, but i'm not sure of the progress.

That's assuming that my hunch was right.  Though, I'm not sure.

drashna, 2019-11-21:
Did you have `EXTRAKEYS_ENABLE = yes` set?  If not, this won't work.

closed 2020-03-20 by stale[bot]

jaqb, 2021-05-10:
I'm experiencing exactly the same issue is as @evantravers on my gmmk pro. Can this be fixed ?

kylejm, 2021-05-13:
I'm having this issue too, exactly as @evantravers is describing. Seems to only happen with media keys :/

Going to have to drop this functionality for now on my board I guess, seems tap dance doesn't play well with media keys.

fauxpark, 2021-05-13:
@jaqb @kylejm have you tried adding `#define TAP_CODE_DELAY 10` to your config.h?

kylejm, 2021-05-13:
@fauxpark didn't work for me :(

Something interesting that happens is this:
1. Single tap the tap-dance key for Skip (which as described by @evantravers, doesn't work)
2. Tap on a switch that I've programmed to be Skip without tap dance (fails, doesn't do anything)
3. Tap on the same switch as 2. again (works!)

This is reproducible 100% of the time for me

jaqb, 2021-05-13:
@fauxpark yes, had that already.

@kylejm I cannot reproduce it, try maybe adding code examples.

What I did figure out just now is that with: `ACTION_TAP_DANCE_DOUBLE(KC_MPRV, KC_MNXT)` you do get repeating `KC_MPRV` when holding that key.

PR #14412 (2021-09-12): Apply `TAP_CODE_DELAY` to Tap Dance key sequences
## Description

This adds a wait before the `unregister_code16` in tap dances, so that it respects the tap code delay for media key compatibility. 

Also, this moves the tap code delay defines to action.h, so that they're picked up most everywhere, rather than just in action.c, in case they're not actually defined.

## Types of Changes

- [x] Core
- [x] Bugfix
- [x] Enhancement/optimization

## Issues Fixed or Closed by This PR

* Fixes #5262  and https://discord.com/channels/440868230475677696/869812721552601088/886583370795065354

## Checklis [... cut]