qmk/qmk_firmware #5301: Dztech/Dz60RGB

race41989, 2019-03-03:
qmk fails to compile . DRIVER_LED_TOTAL  error

## Describe the Bug

It looks like in the latest update something with the Driver_LED_total got shifted or changed causing failure to compile

## System Information

 - Keyboard:  dz60rgd
below is the following error

Compiling: keyboards/dztech/dz60rgb/dz60rgb.c                                                      In file included from quantum/rgb_matrix.h:30:0,
                 from quantum/quantum.h:52,
                 from keyboards/dztech/dz60rgb/dz60rgb.h:3,
                 from keyboards/dztech/dz60rgb/dz60rgb.c:1:
drivers/issi/is31fl3733.h:33:35: error: 'DRIVER_LED_TOTAL' undeclared here (not in a function)
 extern const is31_led g_is31_leds[DRIVER_LED_TOTAL];
                                   ^
 [ERRORS]  
 | 
 | 
 | 
tmk_core/rules.mk:369: recipe for target '.build/obj_dztech_dz60rgb_dzdefault/keyboards/dztech/dz60rgb/dz60rgb.o' failed
make[1]: *** [.build/obj_dztech_dz60rgb_dzdefault/keyboards/dztech/dz60rgb/dz60rgb.o] Error 1
Make finished with errors
Makefile:535: recipe for target 'dztech/dz60rgb:dzdefault' failed
make: *** [dztech/dz60rgb:dzdefault] Error 1

e11i0t23, 2019-03-04:
I can reproduce this error using config.qmk.fm but it compliles fine on PC

The issue appears to be when looking at the source that is produced by config.qmk.fm that it doesn't copy over the config file from the keymap directory which the user keymap is based on, this file is required looking at the source as is where these led defines are set up in

race41989, 2019-03-04:
How would a fix be implemented for this? Is their anything I can do to help?

yanfali, 2019-03-08:
@skullydazed this seems to be an issue on the configurator-api. For whatever reason the define in the config.h isn't getting pulled in correctly. I tested the keymap.c on the CLI and it compiles fine. For some reason it's not including the config.h correctly.

skullydazed, 2019-03-08:
The dztech/dz60rgb board is not setup correctly. It seems there are settings in keymaps/default/config.h that are required for use. These settings need to be in the main dz60rgb config.h.

yanfali, 2019-03-08:
@moyi4681 this change 681db534f3e0e8efa3a9a03ae11b995e4a443ca9 apparently broke configurator. Can you give us some insight why it was moved out of config.h into every keymap please?

e11i0t23, 2019-03-08:
The issue looks to be that they added HHKB Laouts instead of just ISO and both of these layouts have different quantities of LEDS on HHKB layouts 62 where as the non HHKB counterparts seem to only have 61

yanfali, 2019-03-08:
This is abusing keymaps as layouts.

yanfali, 2019-03-08:
```c
diff --git a/keyboards/dztech/dz60rgb/config.h b/keyboards/dztech/dz60rgb/config.h
index 814cb6520..50b26921e 100644
--- a/keyboards/dztech/dz60rgb/config.h
+++ b/keyboards/dztech/dz60rgb/config.h
@@ -32,3 +32,5 @@
 #define DRIVER_ADDR_2 0b1010000 // this is here for compliancy reasons.
 #define DRIVER_COUNT 2
 
+#define DRIVER_1_LED_TOTAL 63
+#define DRIVER_LED_TOTAL DRIVER_1_LED_TOTAL
diff --git a/keyboards/dztech/dz60rgb/keymaps/ansi/config.h b/keyboards/dztech/dz60rgb/keymaps/ansi/config.h
index 2e7122b3c..3ca582e84 100644
--- a/keyboards/dztech/dz60rgb/keymaps/ansi/config.h
+++ b/keyboards/dztech/dz60rgb/keymaps/ansi/config.h
@@ -1,4 +1,6 @@
 #pragma once
 #define dzrgb60_ansi
+#undef DRIVER_1_LED_TOTAL
+#undef DRIVER_LED_TOTAL
 #define DRIVER_1_LED_TOTAL 61
 #define DRIVER_LED_TOTAL DRIVER_1_LED_TOTAL
diff --git a/keyboards/dztech/dz60rgb/keymaps/default/config.h b/keyboards/dztech/dz60rgb/keymaps/default/config.h
deleted file mode 100644
index 558b6023d..000000000
--- a/keyboards/dztech/dz60rgb/keymaps/default/config.h
+++ /dev/null
@@ -1,3 +0,0 @@
-#pragma once
-#define DRIVER_1_LED_TOTAL 63
-#define DRIVER_LED_TOTAL DRIVER_1_LED_TOTAL
diff --git a/keyboards/dztech/dz60rgb/keymaps/hhkb/config.h b/keyboards/dztech/dz60rgb/keymaps/hhkb/config.h
index a0a0e899f..70f4bde25 100644
--- a/keyboards/dztech/dz60rgb/keymaps/hhkb/config.h
+++ b/keyboards/dztech/dz60rgb/keymaps/hhkb/config.h
@@ -1,4 +1,6 @@
 #pragma once
 #define dzrgb60_hhkb
+#undef DRIVER_1_LED_TOTAL
+#undef DRIVER_LED_TOTAL
 #define DRIVER_1_LED_TOTAL 62
 #define DRIVER_LED_TOTAL DRIVER_1_LED_TOTAL
diff --git a/keyboards/dztech/dz60rgb/keymaps/hhkb_iso/config.h b/keyboards/dztech/dz60rgb/keymaps/hhkb_iso/config.h
index bdafd9cd6..2bc3519de 100644
--- a/keyboards/dztech/dz60rgb/keymaps/hhkb_iso/config.h
+++ b/keyboards/dztech/dz60rgb/keymaps/hhkb_iso/config.h
@@ -1,4 +1,6 @@
 #pragma once
 #define dzrgb60_hhkb_iso
+#undef DRIVER_1_LED_TOTAL
+#undef DRIVER_LED_TOTAL
 #define DRIVER_1_LED_TOTAL 62
 #define DRIVER_LED_TOTAL DRIVER_1_LED_TOTAL
diff --git a/keyboards/dztech/dz60rgb/keymaps/iso/config.h b/keyboards/dztech/dz60rgb/keymaps/iso/config.h
index bc830ff18..7ad810a16 100644
--- a/keyboards/dztech/dz60rgb/keymaps/iso/config.h
+++ b/keyboards/dztech/dz60rgb/keymaps/iso/config.h
@@ -1,4 +1,6 @@
 #pragma once
 #define dzrgb60_iso
+#undef DRIVER_1_LED_TOTAL
+#undef DRIVER_LED_TOTAL
 #define DRIVER_1_LED_TOTAL 61
 #define DRIVER_LED_TOTAL DRIVER_1_LED_TOTAL
diff --git a/keyboards/dztech/dz60rgb/keymaps/mekanist/config.h b/keyboards/dztech/dz60rgb/keymaps/mekanist/config.h
index 558b6023d..3e357f202 100644
--- a/keyboards/dztech/dz60rgb/keymaps/mekanist/config.h
+++ b/keyboards/dztech/dz60rgb/keymaps/mekanist/config.h
@@ -1,3 +1,5 @@
 #pragma once
+#undef DRIVER_1_LED_TOTAL
+#undef DRIVER_LED_TOTAL
 #define DRIVER_1_LED_TOTAL 63
 #define DRIVER_LED_TOTAL DRIVER_1_LED_TOTAL
```
Is enough to fix it.

yanfali, 2019-03-11:
@race41989 can you confirm this is fixed and close the issue please? Thanks

drashna, 2019-04-26:
Is this still an issue?

noroadsleft, 2019-11-21:
I believe this was solved by #7223.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc