qmk/qmk_firmware #5305: Preonic Rev3: Neigher LGUI nor RGUI works

hide-tanaka, 2019-03-04:
## Describe the Bug

I checked out the master, built preonic rev3 default, and flash it to my preonic rev 3 keyboard with QMK Toolbox. Then, I found the KC_LGUI doesn't work. It doesn't generate a key event at all. The same issue occurs with KC_RGUI. Oddly enough, no matter which key I assigned the KC_LGUI or KC_RGUI to, it doesn't work. No key event is generated.

## System Information

 - Keyboard:  Preonic
   - Revision (if applicable):  rev3
 - Operating system: macOS Mojave
 - AVR GCC version: 7.3.0

 - ARM GCC version: 7.3.1

 - QMK Firmware version: 0.6.305

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [√ ] Karabiner
   - [ ] Other:

## Additional Context

fauxpark, 2019-03-04:
Try unplugging your keyboard, then holding Left GUI as you plug it back in.

The default keymap for the Preonic Rev3 will have [Bootmagic](https://docs.qmk.fm/#/feature_bootmagic) turned on, which allows you to change some config options when the keyboard is starting up. It's possible you somehow accidentally activated the GUI key lock function.

hide-tanaka, 2019-03-05:
Thanks! I was able to get it to work.

closed 2019-03-05 by hide-tanaka