qmk/qmk_firmware #5346: Tap dancing timeout control

zored, 2019-03-08:
## Summary
Allow different `TAPPING_TERM` for each tap.

## Description
I want to hold Z for Ctrl and double-hold for Ctrl-Shift.
- Fast Ctrl is prefferred, but with low `TAPPING_TERM` it's almost impossible to double-hold!
- With higher `TAPPING_TERM` I can double-hold, but I need to wait significant time for Ctrl.

Are there any ways to control these tap dancing timeouts? I tried to modify `qk_tap_dance_state_t.timer` inside of `user_fn_on_each_tap` callback, but with no success now... 😊

P.S.: Thank you for great firmware! ❤️

drashna, 2019-03-08:
Yes and no. 

There is this: https://github.com/qmk/qmk_firmware/pull/5009?w=1

This should allow you to set the tapping term per keycode, so that each keycode can use a different tapping term.  

But for different tapping terms for a single key?  That's a lot more complex.

zored, 2019-03-10:
@drashna thanks!

First I tried to hack timer, but it seems that simple `+=100;` is not enough for my timer implementation.

So I implemented simple dynamic timer control. There is basic implementation of it [living in my repo](https://github.com/qmk/qmk_firmware/commit/0998498778cfc324a00bd3d6953927770f07fa26#diff-e90382315ce33b080b191cd608f933b2R192) 😊

zored, 2019-03-10:
So it seems that I fixed slow tap dance modifiers with custom interruption check. What do you think, is it a good solution to document it? 🤔

drashna, 2019-03-18:
Yeah, I think it would be. :)

closed 2019-11-21 by drashna