qmk/qmk_firmware #5348: make preonic/rev3 doesn't work as expected

atrnh, 2019-03-08:
## Describe the Bug
Following the example provided in the [How to Compile section of the docs](https://docs.qmk.fm/#/?id=how-to-compile), I'd expect to be able to compile a new firmware by running

```
make preonic/rev3:<keymap>
```
...but it causes a `make` error:
```
tmk_core/chibios.mk:89: lib/chibios/os/hal/lib/streams/streams.mk: No such file or directory
make[1]: *** No rule to make target 'lib/chibios/os/hal/lib/streams/streams.mk'.  Stop.
```

## System Information

 - Keyboard:  Preonic
   - Revision (if applicable):  rev3
 - Operating system: Vagrant
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version: 0.6.300

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

To fix, I changed
https://github.com/qmk/qmk_firmware/blob/fef206f661cc5d572549e91faa2d628493314510/keyboards/preonic/rules.mk#L76

to `DEFAULT_FOLDER = preonic/rev3` and ran

```
make preonic:<keymap>
```

In #4950 it appears that running this command works:

```
make preonic/v3:custom:dfu-util
```

Apologies if this has already been patched, I haven't git-pulled in a while

drashna, 2019-03-18:
You need to run `make git-submodule`, as it depends on the ChibiOS stuff, which is a submodule, and not downloaded by default.

atrnh, 2019-03-18:
@drashna I've already done that

closed 2019-03-18 by atrnh