qmk/qmk_firmware #5352: Cant compile hex for my dz60

Aggeloz, 2019-03-09:
Every time i try to compile msys2 gives me an error that says : 

Linking: .build/dz60_Aggelos.elf                                                                    [ERRORS]
 |
 | .build/obj_dz60_Aggelos/quantum/keymap_common.o: In function `keymap_key_to_keycode':
 | C:\Users\Aggelos\Desktop\qmk_firmware/quantum/keymap_common.c:212: undefined reference to `keymaps'
 | C:\Users\Aggelos\Desktop\qmk_firmware/quantum/keymap_common.c:212: undefined reference to `keymaps'
 | collect2.exe: error: ld returned 1 exit status

mechmerlin, 2019-03-10:
could you give us more info as to what command you are using to compile?

Aggeloz, 2019-03-10:
Omg im sorry. I fixed it by remaking my own layout and forgot to close my issue.

closed 2019-03-10 by Aggeloz