qmk/qmk_firmware #5361: KBD19X info.json numpad and arrows incorrect

beadlock, 2019-03-10:
When using QMK configurator to make a layout I found that my numpad and arrow cluster were registering keys incorrectly. 

Keys affected: 
Up, Down, Left, Right arrows. 
1,2,3 Enter, 0, Decimal on numpad

After flashing a .hex to the keyboard, the keys affected will register as other keys. For example, 1 registers as 2, 2 as 3, enter as up arrow. 0 as down arrow, left arrow as 0 etc... 

Yanfali helped and provided a default .hex that worked correctly. (https://cdn.discordapp.com/attachments/473506116718952450/554386248530395178/kbdfans_kbd19x_default.hex)

- [ ] Add default keymap to QMK configurator
- [ ] Correct issue with info.json

saesh, 2019-10-10:
This has been resolved in #5362 and can be closed.

The default keymap produced by the configurator is the same as the default keymap in the kbd19x keyboard directory which works fine.

noroadsleft, 2019-10-10:
@saesh Thanks for flagging this as resolved.

closed 2019-10-10 by noroadsleft