qmk/qmk_firmware #5369: ld link Error: Conflicting CPU architectures 13/1

kb3dow, 2019-03-11:
Compiling for the Infinity60 keyboard gives a linking error saying 
Conflicting CPU architectures.
%  make infinity60:default
gives at the end ...
Assembling: lib/chibios/os/common/ports/ARMCMx/compilers/GCC/chcoreasm_v7m.S                        [OK]
Linking: .build/infinity60_default.elf                                                              [ERRORS]
 |
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/crt0.o: Conflicting CPU architectures 13/1
 | /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/crt0.o
 | collect2: error: ld returned 1 exit status
 |
tmk_core/rules.mk:290: recipe for target '.build/infinity60_default.elf' failed
make[1]: *** [.build/infinity60_default.elf] Error 1
Make finished with errors
Makefile:535: recipe for target 'infinity60:default' failed
make: *** [infinity60:default] Error 1

sebirdman, 2019-03-12:
I had the same issue:

Sounds like this is a bug with ubuntu: https://github.com/bbcmicrobit/micropython/issues/514#issuecomment-404759614 

The fix there seems to have fixed it here as well.

kb3dow, 2019-03-12:
> Sounds like this is a bug with ubuntu: bbcmicrobit/micropython#514
> (comment)

Great, thanks. I'll try it out.

kb3dow, 2019-03-12:
> 
> 
> I had the same issue:
> 
> Sounds like this is a bug with ubuntu: [bbcmicrobit/micropython#514 (comment)](https://github.com/bbcmicrobit/micropython/issues/514#issuecomment-404759614)
> 
> The fix there seems to have fixed it here as well.

That fixes the issue.

closed 2019-03-12 by kb3dow