qmk/qmk_firmware #5411: DZtech/DZ60RGB | Keyboard does not register multiple keys simultaneously

MatthewRobo, 2019-03-15:
## Describe the Bug
On my Massdrop CTRL, when I press four keys (e.g. A, S, D, F) at once in a textbox they will all show up at once. When I do the same with my DZ60RGB the four keys will appear one at a time, instead of all showing up in a single frame.
This issue occurs even with RGB disabled.
This issue occurs even with the following lines:
```c
//matthewrobo/config.h
#define FORCE_NKRO
#undef  RGB_MATRIX_SKIP_FRAMES 
#define RGB_MATRIX_SKIP_FRAMES 10
#define QMK_KEYS_PER_SCAN 4
```

RGB Matrix processing should be disabled as the following lines are commented out:
```c
//quantum.c
  #ifdef RGB_MATRIX_ENABLE
    // rgb_matrix_task();
    if (rgb_matrix_task_counter == 0) {
      //rgb_matrix_update_pwm_buffers();
    }
    rgb_matrix_task_counter = ((rgb_matrix_task_counter + 1) % (RGB_MATRIX_SKIP_FRAMES + 1));
  #endif
```

## System Information

 - Keyboard: DZ60RGB
   - Revision (if applicable): N/A
 - Operating system: Windows 10 Home 1803
 - AVR GCC version: (AVR_8_bit_GNU_Toolchain_3.6.1_1752) 5.4.0

 - ARM GCC version: 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 - QMK Firmware version: 0.6.246 (xscorpion2/qmk_firmware:dev/rgb_matrix_overhaul)

 - Any keyboard related software installed? 
   - [X] AutoHotKey (not running)
   - [ ] Karabiner
   - [ ] Other:

## Additional Context
Keyboard has Gateron Yellows (plate-mounted SMD LEDs compatible).

yanfali, 2019-03-15:
Can you test it on another operating system too please. Just want to verify it's not related to your windows install.

drashna, 2019-03-15:
Unfortunately, I suspect that this is something to do with the DZ60RGB implementation specifically.

I have a Planck Rev6 (mono), and I cannot replicate this issue.
I mention it, because it's using the same STM32F303xC chip in both boards, IIRC.  Though, how the ARM stuff is implemented is slightly different so that may be why.

slamjamm, 2019-03-19:
Same issue with the xd75. No issues with a different, non QMK board on this computer. Are there any firmwares without this problem that could be flashed to test if the problem is with the board or with the original firmware?

drashna, 2019-03-19:
@slamjamm Unfortunately, each keyboard's firmware is very specific to that keyboard.  There is no safe way to flash another board's firmware.

That said, the best way is to roll back until you find the change in the firmware that breaks it.  

Though, I don't think that this is a MCU/controller related issue, as I'm I can verify that NKRO works on the Planck Rev6.  Though, it may be a configuration setting for the MCU that is causing this.

MrShiister, 2019-04-19:
Just to chime in, I have a DZ60 and been having the same issues for many months. Posted the same question on reddit [HERE](https://www.reddit.com/r/MechanicalKeyboards/comments/b23way/dz60_using_qmk_pressing_multiple_keys_at_the_same/)

drashna, 2019-04-26:
@MatthewRobo Is this still an issue?

MatthewRobo, 2019-04-26:
@drashna Just updated and yeah, still an issue.

drashna, 2019-04-26:
Okay.

NKRO enabled, right? Or just 6KRO? 

Also, I'm curious what EX Switch Hitter shows.  For me, this was my output: 
```
09:25.0908 R (0x52, BIOS 0x13) DOWN
09:25.0914 S (0x53, BIOS 0x1F) DOWN
09:25.0919 T (0x54, BIOS 0x14) DOWN
09:25.0927 D (0x44, BIOS 0x20) DOWN
09:26.0056 R (0x52, BIOS 0x13) UP -> 152ms
09:26.0063 S (0x53, BIOS 0x1F) UP -> 149ms
09:26.0071 T (0x54, BIOS 0x14) UP -> 152ms
09:26.0079 D (0x44, BIOS 0x20) UP -> 152ms
```

MatthewRobo, 2019-04-26:
NKRO enabled. [(video)](https://streamable.com/g5l5s)

```
14:57.0514 5 (0x35, BIOS 0x06) DOWN
14:57.0522 6 (0x36, BIOS 0x07) DOWN
14:57.0530 7 (0x37, BIOS 0x08) DOWN
14:57.0538 8 (0x38, BIOS 0x09) DOWN
14:57.0546 9 (0x39, BIOS 0x0A) DOWN
14:57.0554 T (0x54, BIOS 0x14) DOWN
14:57.0562 Y (0x59, BIOS 0x15) DOWN
14:57.0570 I (0x49, BIOS 0x17) DOWN
14:57.0578 G (0x47, BIOS 0x22) DOWN
14:57.0586 H (0x48, BIOS 0x23) DOWN
14:57.0594 K (0x4B, BIOS 0x25) DOWN
14:57.0602 V (0x56, BIOS 0x2F) DOWN
14:57.0610 B (0x42, BIOS 0x30) DOWN
14:57.0618 N (0x4E, BIOS 0x31) DOWN
14:57.0626 M (0x4D, BIOS 0x32) DOWN
14:57.0634 , (0xBC, BIOS 0x33) DOWN
14:57.0642 Space (0x20, BIOS 0x39) DOWN
14:57.0650 U (0x55, BIOS 0x16) DOWN
14:58.0058 J (0x4A, BIOS 0x24) DOWN
14:58.0418 J (0x4A, BIOS 0x24) UP -> 360ms
14:58.0426 5 (0x35, BIOS 0x06) UP -> 912ms
14:58.0474 6 (0x36, BIOS 0x07) UP -> 952ms
14:58.0482 7 (0x37, BIOS 0x08) UP -> 952ms
14:58.0490 8 (0x38, BIOS 0x09) UP -> 952ms
14:58.0498 9 (0x39, BIOS 0x0A) UP -> 952ms
14:58.0506 T (0x54, BIOS 0x14) UP -> 952ms
14:58.0514 Y (0x59, BIOS 0x15) UP -> 952ms
14:58.0522 U (0x55, BIOS 0x16) UP -> 872ms
14:58.0530 I (0x49, BIOS 0x17) UP -> 960ms
14:58.0538 G (0x47, BIOS 0x22) UP -> 960ms
14:58.0546 H (0x48, BIOS 0x23) UP -> 960ms
14:58.0554 K (0x4B, BIOS 0x25) UP -> 960ms
14:58.0562 V (0x56, BIOS 0x2F) UP -> 960ms
14:58.0570 B (0x42, BIOS 0x30) UP -> 960ms
14:58.0578 N (0x4E, BIOS 0x31) UP -> 960ms
14:58.0586 M (0x4D, BIOS 0x32) UP -> 960ms
14:58.0594 , (0xBC, BIOS 0x33) UP -> 960ms
14:58.0602 Space (0x20, BIOS 0x39) UP -> 960ms
```

```
16:11.0787 S (0x53, BIOS 0x1F) DOWN
16:11.0794 D (0x44, BIOS 0x20) DOWN
16:11.0802 F (0x46, BIOS 0x21) DOWN
16:11.0810 G (0x47, BIOS 0x22) DOWN
16:11.0906 S (0x53, BIOS 0x1F) UP -> 120ms
16:11.0914 D (0x44, BIOS 0x20) UP -> 120ms
16:11.0923 F (0x46, BIOS 0x21) UP -> 120ms
16:11.0930 G (0x47, BIOS 0x22) UP -> 120ms

```

rayno206, 2019-05-13:
is this still an issue? or fixed? any news on the fixes guys?

XScorpion2, 2019-05-13:
I remember someone found which pr caused the issue, I think it was related to usb descriptor or endpoint changes. I think MathewRobo was still having issues last I heard.

rayno206, 2019-05-13:
I have the same issue with this pcb, can't force it into nkro.

XScorpion2, 2019-05-13:
39bd760faf2666e91d6dc5b199f02fa3206c6acd
That was the commit that XaeL reported in discord to have caused a similar problem to this. Not sure if it ever got fixed, asked in discord about this.

rayno206, 2019-05-13:
ok, thanks for your help, I will try to seek help from QMK's discord too.

rayno206, 2019-05-17:
I tried it, but still none worked so far :(

rayno206, 2019-05-23:
I just got confirmation from one of the guy from KBDFans that the dz60rgb doesn't support NKRO, so it won't do anything even if we try to turn on the force NKRO

fauxpark, 2019-05-24:
I see [diodes on the PCB](https://cdn.shopify.com/s/files/1/1473/3902/products/5_c4fdf10f-f3b8-4bd7-8d67-026ac3067ab3_1800x1800.jpg?v=1548735125), so in theory NKRO should work.

MatthewRobo, 2019-05-24:
I have no idea where all this nonsense about it not supporting NKRO is coming from because [I'm very sure it does](https://i.imgur.com/UQsDbCp.png).

rayno206, 2019-05-24:
well, obviously it's not coming from me though, I just pm them to see if they are willing to try and fix this, and I hope this board can be fixed for the NKRO as well. But that's what they told me though

rayno206, 2019-06-16:
I believe this issue has been fixed with the release 0.6.400 right?

drashna, 2019-06-17:
Is this still happening?

rayno206, 2019-06-17:
I flash 0.6.400 firmware then I enabled nkro and bootmagic in rule.mk file, and I tried space + N when plugging in the keyboard, and it worked, NKRO worked for me.

MatthewRobo, 2019-06-17:
NKRO always worked for me @rayno206.
The issue is still occurring @drashna.
[Here](https://github.com/qmk/qmk_firmware/files/3296565/00001.log) is the Switch Hitter log.

rayno206, 2019-06-17:
ah sht, I got what you meant now, sr @MatthewRobo, I wasn't able to enable the NKRO ever before, and I thought this issue is about NKRO. I notice it now, yeah you are right, they won't register the same time even though we press them at same time.

PatzMatias, 2019-11-18:
@MatthewRobo Is this still an issue? I currently have this board and would like to have this feature working as well. I noticed on your keymap's `config.h` that you already have `FORCE_NKRO` and `rules.mk` have `NKRO_ENABLE = yes`, does any of these configs help enable NKRO?

rayno206, 2019-11-18:
@PatzMatias put `BootMagic=full` and `NKRO_ENABLE = yes` in `rule.mk`. Then use your board in bootmagic mode by holding down space+B while plugging in your board, this should enable NKRO for you as it worked for me.

rayno206, 2019-11-18:
@PatzMatias However, I believe this is the issue about the keys are not registered at the same time when pressed (something wrong about the timing of the keys pressed)

PatzMatias, 2019-11-18:
@rayno206 Thanks! Do I have to use @MatthewRobo 's keymap or just using those settings should be good to go? Sorry, I just assembled my first mechanical keyboard yesterday and currently trying to get to know the issues and features DZ60RGB has.

rayno206, 2019-11-18:
@PatzMatias You can just use those settings, I just enabled those settings in the `rule.mk` file when compiling firmware and use the board in BootMagic mode, that's all

MatthewRobo, 2019-11-18:
@PatzMatias This issue was actually solved for me due to #6668. 
On the DZ65RGB (which had similar behavior), setting the `#define USB_POLLING_INTERVAL_MS 10` to `#define USB_POLLING_INTERVAL_MS 1` made the keys activate simultaneously, making my DZ65RGB finally viable for playing fighting games. The only issue is that it seems to chatter more than it did before.
Since I don't use my DZ60RGB anymore, please tell me if this works for you too.

rayno206, 2019-11-18:
@MatthewRobo May I ask where did you change this setting? in the tmk_core directory?

rayno206, 2019-11-18:
@MatthewRobo Never mind, I got it, thanks a lot!

MatthewRobo, 2019-11-18:
@rayno206 in my `config.h`.

PatzMatias, 2019-11-20:
@rayno206 Thank you for the tip. It did activate NKRO. @MatthewRobo I've changed USB_POLLING_INTERVAL_MS to 1 but i'm not sure how to tell if it worked. Here's a log from switch hitter. I was not able to record a log before doing so. But thanks as well.

`13:08.0520 A (0x41, BIOS 0x1E) DOWN
13:08.0524 D (0x44, BIOS 0x20) DOWN
13:08.0528 S (0x53, BIOS 0x1F) DOWN
13:08.0530 F (0x46, BIOS 0x21) DOWN
13:08.0533 J (0x4A, BIOS 0x24) DOWN
13:08.0535 ; (0xBA, BIOS 0x27) DOWN
13:08.0542 K (0x4B, BIOS 0x25) DOWN
13:08.0545 L (0x4C, BIOS 0x26) DOWN
13:08.0672 S (0x53, BIOS 0x1F) UP -> 145ms
13:08.0687 A (0x41, BIOS 0x1E) UP -> 168ms
13:08.0690 D (0x44, BIOS 0x20) UP -> 167ms
13:08.0693 K (0x4B, BIOS 0x25) UP -> 150ms
13:08.0696 L (0x4C, BIOS 0x26) UP -> 151ms
13:08.0698 ; (0xBA, BIOS 0x27) UP -> 164ms
13:08.0703 F (0x46, BIOS 0x21) UP -> 174ms
13:08.0706 J (0x4A, BIOS 0x24) UP -> 173ms
13:09.0651 J (0x4A, BIOS 0x24) DOWN
13:09.0655 L (0x4C, BIOS 0x26) DOWN
13:09.0657 ; (0xBA, BIOS 0x27) DOWN
13:09.0661 K (0x4B, BIOS 0x25) DOWN
13:09.0669 A (0x41, BIOS 0x1E) DOWN
13:09.0672 S (0x53, BIOS 0x1F) DOWN
13:09.0674 D (0x44, BIOS 0x20) DOWN
13:09.0676 F (0x46, BIOS 0x21) DOWN
13:09.0744 ; (0xBA, BIOS 0x27) UP -> 88ms
13:09.0768 K (0x4B, BIOS 0x25) UP -> 107ms
13:09.0771 L (0x4C, BIOS 0x26) UP -> 116ms
13:09.0775 J (0x4A, BIOS 0x24) UP -> 124ms
13:09.0780 S (0x53, BIOS 0x1F) UP -> 109ms
13:09.0790 A (0x41, BIOS 0x1E) UP -> 121ms
13:09.0793 D (0x44, BIOS 0x20) UP -> 119ms
13:09.0796 F (0x46, BIOS 0x21) UP -> 120ms `

closed 2020-03-19 by stale[bot]