qmk/qmk_firmware #5415: Honeycomb spamming input after flashing

filoxo, 2019-03-16:
Honeycomb keymap is spamming input with the entire keymap after flashing the pro micro, unplugging and plugging it back in; it sends all the keycodes repeated quickly. Verified on 3 separate pro micros. 

Sample output:

```sh
dev/qmk_firmware [master●] » deeeeeeeeeeeeeeeeeeffffffffffffffffffaaaaaaaaaaaaaaaaaabbbbbbbbbbbbbbbbbb555555555555555555555555555555555555555555555555555555555555
```

## System Information

 - Keyboard:  Honeycomb
   - Revision (if applicable):  
 - Operating system: Mac OS X
 - AVR GCC version: 7.3.0
 - ARM GCC version: N/A
 - QMK Firmware version: 0.6.262

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

It uses a custom matrix to communicate to the BT module using UART. My current guess is that its conflicting with some recent update to QMK internals. 

NOTE: I was the original author of the Honeycomb so I'm going to try to triage this as possible.

filoxo, 2019-03-16:
Sample console output when connected: 

```
*** Keyhive - Honeycomb Macropad connected -- 0xFEED:0xACC8
  > .
    Time out in keyboard.
    [truncated 234 lines] 
    Time out in keyboard.
    Requested packet, data 3 was 0
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    Requested packet, data 3 was 255
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    0x737373
    r/c 0123456789ABCDEF
    00: 0111001101110011
    keyboard_report: 00 00 07 00 00 00 00 00 
    
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    Requested packet, data 3 was 115
    r/c 0123456789ABCDEF
    00: 0111001101110011
    keyboard_report: 00 00 07 08 00 00 00 00 
    
    Time out in keyboard.
  > Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    0x737373
    r/c 0123456789ABCDEF
    00: 0111001101110011
    keyboard_report: 00 00 07 08 09 00 00 00 
  > 
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    0x737373
    r/c 0123456789ABCDEF
    00: 0111001101110011
    keyboard_report: 00 00 07 08 09 04 00 00 
  > 
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    0x737373
    r/c 0123456789ABCDEF
    00: 0111001101110011
    keyboard_report: 00 00 07 08 09 04 05 00 
  > 
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    0x737373
    r/c 0123456789ABCDEF
    00: 0111001101110011
    keyboard_report: 00 00 07 08 09 04 05 22 
    
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    0x737373
    r/c 0123456789ABCDEF
    00: 0111001101110011
    keyboard_report: 00 00 07 08 09 04 05 22 
    
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    0x737373
    r/c 0123456789ABCDEF
    00: 0111001101110011
    keyboard_report: 00 00 07 08 09 04 05 22 
    
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    0x737373
    r/c 0123456789ABCDEF
    00: 0111001101110011
    keyboard_report: 01 00 07 08 09 04 05 22 
    keyboard_report: 01 00 07 08 09 04 05 22 
    keyboard_report: 01 00 07 08 09 04 05 22 
  > keyboard_report: 00 00 07 08 09 04 05 22 
    
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    Time out in keyboard.
    0x737373
*** Keyhive - Honeycomb Macropad disconnected -- 0xFEED:0xACC8
```

drashna, 2019-03-19:
My guess is that there is a short somewhere. Either a PCB issue, or some stray solder bridging something.  

And that 3 separate Pro Micros exhibited this .... really points to this.

Though, have you flashed just the Pro Micro, when not connected to the board?  And if so, does it exhibit this behavior too?

filoxo, 2019-03-19:
The user who reported this (klackygears), the designer of the PCB (snipeye), and myself have been trying to diagnose this over on Discord. Here's a quick recap:

- the chatter input is expected if pro micro is not connected to BT module due to UART requiring a connection, else the pins will receive random input from environmental conductivity (I think that's worded right... but I'm not familiar with UART)
- klackygears seems to have good solder connections, and reviewed Snipeye's Dichotomy post on troubleshooting the module
- I've reverted to an older version of QMK to try compiling from that, and klackygears has reported that it's still doing the same thing
- I'm going to be receiving another hardware kit to try to replicate the issue as well as validate a fix
- Current theory is that the NRF module is bad or just needs to be reprogrammed (which would mean that the issue is unrelated to QMK)

filoxo, 2019-03-27:
OK, I finally got around to being able to test. I'm still getting this same issue. I'm out of ideas other than it has to do with some underlying compiler or submodule change.

filoxo, 2019-08-13:
Found the issue. The user had soldered the voltage level shifter in the incorrect orientation, so desoldering and reorienting fixes the problem. Closing.

closed 2019-08-13 by filoxo