qmk/qmk_firmware #5416: A report on hardware TWI/I2C AVR drivers in QMK

datagrok, 2019-03-16:
## Feature Request Type

- [x] Core functionality
- [x] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

There are many files in qmk_firmware that implement drivers for the Two-Wire Interface (TWI) hardware feature in AVR chips (also known as I2C or I²C). I am learning how to use this feature to mod a peripheral onto my own keyboard and I wondered:

1. What are the differences between all of these files?
1. Is there a "best" and "latest" implementation?
2. If the drivers are copied from other locations, who is the original author? What is the license? Is there an upstream repository?
3. Would a PR that refactors these various implementations into a limited set of top-level libraries (included by way of `git submodule` where possible) benefit the QMK project?

A more detailed report follows.

---

As of commit 2558372b3, there are 55 separate files containing I2C drivers. Among these, only 15 are unique.

```
$ grep -rl 'TW[SB]R\s*=' | grep -v '^lib/\(ugfx\|chibios\)' | grep -v 'issi' | xargs md5sum | sort | awk '{print $2, $1}' | uniq -f1 --group | cut -f1 -d' '
```

These may be further partitioned into minor revisions of 6 distinct implementations:

- https://github.com/abcminiuser/lufa
  - lib/lufa/LUFA/Drivers/Peripheral/AVR8/TWI_AVR8.h
- https://github.com/c-herring/AVRTWIlib
  - keyboards/meira/TWIlib.c
  - keyboards/lfkeyboards/TWIlib.c
- https://github.com/chrisramsay/i2cmaster (unofficial; [origin](http://homepage.hispeed.ch/peterfleury/avr-software.html))
  - keyboards/ergodone/twimaster.c
  - keyboards/sx60/twimaster.c
  - keyboards/handwired/dactyl/twimaster.c
- https://github.com/g4lvanix/I2C-master-lib
  - drivers/avr/i2c_master.c
- `i2c.c`, can't find canonical author or upstream repo
  - keyboards/comet46/i2c.c
  - keyboards/vitamins_included/i2c.c
  - keyboards/hadron/i2c.c
  - keyboards/deltasplit75/i2c.c
  - keyboards/dc01/left/i2c.c
- `i2c.c` by @luizribeiro; is qmk_firmware the origin?
  - keyboards/bfake/i2c.c
  - keyboards/panc60/i2c.c
  - keyboards/jj40/i2c.c

With only two exceptions, the differences between versions within implementations are trivial changes to comments or whitespace, so could fairly easily be collapsed into a single revision or replaced with the upstream version. The exceptions were:

An indication of a possible bug (fix) between versions:

`$ diff -b keyboards/meira/TWIlib.c keyboards/lfkeyboards/TWIlib.c`
```diff
40d37
< 		            break;
```

The modifications made for the keyboard `vitamins_included` seems to add a feature:

`$ diff -b keyboards/comet46/i2c.c keyboards/vitamins_included/i2c.c`
```diff
160a161
>   contacted_by_master = true;
```

pelrun, 2019-03-16:
If you're making anything new, use `drivers/avr/i2c_master.c`. It's the one that will be kept maintained, and shares a common API with the ARM driver, simplifying ports.

datagrok, 2019-03-17:
@pelrun thanks!

closed 2019-03-17 by datagrok