qmk/qmk_firmware #5425: BL_TOGG is not behaving as expected

HMarxen, 2019-03-17:
Hi, I’ve put BL_TOGG in my keymap and I was expecting that it would toggle between backlight off and the last set backlight brightness. However, all it does is if the backlight is off it will turn to the lowest brightness setting. If the backlight is already on at any brightness setting it does nothing. Am I doing something wrong or is this the intended behaviour?

drashna, 2019-03-19:
Which keyboard is this?

HMarxen, 2019-03-19:
A xd75AM. This keyboard doesen't have the the Backlight on a hardware PWM pin. Might the software PWM be the problem?

FSund, 2019-05-21:
I'm having the same issue with a AMJ66 PCB that uses D4 as the backlight pin.

*  `BL_TOGG` turns on the backlight, but not off again
* `BL_ON` can not be used to turn it on, but will put it at full brightness if it's below that
* `BL_OFF` does nothing
* `BL_STEP` steps between different brightnesses, and sometimes enables breathing
* `BL_INC` and `BL_DEC` does nothing
* `BL_BRTG` does nothing

zvecr, 2019-05-21:
Have you tried setting `#define BACKLIGHT_ON_STATE 1` in the boards config.h? I had to do so on the staryu to get the correct toggle behaviour https://github.com/qmk/qmk_firmware/blob/0aece4ddab37e603699cb4ba9e8b72fdaf5a619c/keyboards/staryu/config.h#L68.

FSund, 2019-05-22:
Tried `#define BACKLIGHT_ON_STATE 1` now, but it only seemed to make things worse.

drashna, 2019-06-17:
@HMarxen @FSund Have you tried using the PR referenced right above this comment?  

And if so, does that help?

FSund, 2019-06-17:
@drashna Yes, it works perfectly as long as I remember `#define BACKLIGHT_ON_STATE 1`. I left a comment in the PR about it a while back.

fauxpark, 2019-06-17:
Thanks for reminding me to document `BACKLIGHT_ON_STATE` 😆 done in #5983

FSund, 2019-08-08:
I think this can be closed now? Works fine on my board after https://github.com/qmk/qmk_firmware/pull/5983 and https://github.com/qmk/qmk_firmware/pull/5945 were merged.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc