qmk/qmk_firmware #5436: Identifying support for qmk firmware.

AK2427, 2019-03-18:
How can I identify that my keyboard supports qmk or any other firmware? I have been searching for it on the internet but haven't been able to find anything. Please tell me any way that I could identify that my keyboard has support for any firmware.

Thanks

noroadsleft, 2019-03-18:
This section of the Docs may be helpful to you:

- [Hardware - QMK Firmware](https://docs.qmk.fm/#/hardware)

What keyboard is it?

AK2427, 2019-03-19:
Keyboard 
https://www.amazon.in/Circle-Squadron-Professional-Keyboard-Black/dp/B07DJBTJHX

drashna, 2019-03-19:
Most likely not. 

A lot of OEM keyboards like this don't have a way to flash the keyboard. Also, the MCU (controller) may be locked to prevent flashing, actually.  

But basically, you'd need to figure out what controller it uses, see if it's even supported, and then basically trace the matrix. 

The other way is to see about getting/making the hasu USB to USB converter, which allows you to add QMK support to any keyboard.

fauxpark, 2019-03-19:
> the MCU (controller) may be locked to prevent flashing

Not locked so much as they use an MCU that isn't AVR or ARM. You'll often find OEM boards like the WASD V2 have a chip designed specifically for use in keyboards, which is OTP (one-time programmable) and not very powerful anyway.

drashna, 2019-03-19:
That's very true, as well.  Though, a lot of the per key rgb boards tend to be ARM.  Corsair uses ARM, IIRC. 

But yeah, programming the keyboard directly like that.... it's not too likely. :(

AK2427, 2019-03-19:
Initially I thought about buying the hasu USB to USB converter but it was always out of stock so my best bet was to programme the keyboard itself or just buy a new PCB but, as it's my only mechanical keyboard and my parents won't buy me a new one so I refrain myself from changing the PCB

AK2427, 2019-03-19:
> Initially I thought about buying the hasu USB to USB converter but it was always out of stock so my best bet was to programme the keyboard itself or just buy a new PCB but, as it's my only mechanical keyboard and my parents won't buy me a new one so I refrain myself from changing the PCB

noroadsleft, 2019-03-19:
Not sure about the RGB, but the Hasu USB-to-USB Converter can definitely run QMK.

https://github.com/qmk/qmk_firmware/tree/master/keyboards/converter/usb_usb

drashna, 2019-03-28:
^  Also, you can get them here: https://www.1upkeyboards.com/shop/controllers/usb-to-usb-converter/

Additionally, you could build one yourself, if you source the parts (not difficult).

closed 2019-11-21 by drashna