qmk/qmk_firmware #5439: Adafruit feather 32u4 Bluetooth support not working.

harshitgoel96, 2019-03-18:
A few months ago I used QMK on Adafruit feather 32u4 and was able to get complete functional bluetooth keyboard working. Last night after reflashing the code, the Bluetooth output only works till USB is connected to the feather. When I remove the usb it does not work over bluetooth at all.

The feather is connecting to mobile, but there is no bt output unless usb is connected. I have tried OUT_BT and OUT_AUTO, but nothing seems to be working.

drashna, 2019-03-19:
This may be related to: https://github.com/qmk/qmk_firmware/pull/4131

But I'm not entirely sure about that.

Do you know if there is a version that does work correctly?

harshitgoel96, 2019-03-24:
the last time I built the firmware, with working Bluetooth was when I posted issue  #2229

drashna, 2019-03-29:
Just thinking about it, this change may be why: https://github.com/qmk/qmk_firmware/pull/3951?w=1

Unfortunately, I don't have a BT controller to test out, ATM... so I can't test this out myself.... but if you could figure out what change caused this specifically, it would be awesome, and help us to fix the issue.

harshitgoel96, 2019-03-31:
@drashna I don't have skills to do it. I don't have much knowledge about embedded systems :(

deltapoot, 2019-03-31:
Can you downgrade the firmware on the adafruit feather to make this to work?

harshitgoel96, 2019-03-31:
@deltapoot I dont think the issue is with Adafriut Firmware itself, but with QMK. I do not know which version of qmk is going to work either.

deltapoot, 2019-03-31:
I see, I understand now. Thank you.

fcoury, 2019-04-23:
Just to add to the conversation, I have reverted to commit 

https://github.com/qmk/qmk_firmware/commit/46cf8cc9b33a3c8bdbb0ce7df7f48f28e4d979a5

Which is the parent of:

https://github.com/qmk/qmk_firmware/commit/39bd760faf2666e91d6dc5b199f02fa3206c6acd

Which is the commit resulting from the mentioned PR and I still have the same issue described here.

fcoury, 2019-04-23:
OK, after a ton of investigation I have made it work with the following code changes:

- Setting the keyboard to use `OUTPUT_USB` by default
- Removing bluetooth initialization using `adafruit_ble_task()` from `lufa.c``main()` and moving it to be called lazily when the output is toggled to Bluetooth on `quantum.c`
- Removing buffered output of keystrokes (via `RingBuffer`), sending it directly instead

And then following this procedure:

1. Connect to keyboard via bluetooth
2. Change output to Bluetooth via `OUT_BT` keypress

I will continue this investigation and will report back when I know more.

marhalloweenvt, 2019-05-15:
Hey @fcoury,
Can i use your qmk_firmware branch fcoury/fix/adafruit_ble to compile a bluetooth fw for Adafruit feather 32u4? I am doing a handwired keyboard with it.

harshitgoel96, 2019-05-19:
paging @wez who originally wrote the code for feather.

Hi Wez, there seems to be some issue.

wez, 2019-05-19:
Sorry; I haven't been following qmk since I contributed that patch, and I don't have time to review all of the changes since that time.

fcoury, 2019-08-26:
@wez it seems like it's not something on QMK that is causing the issue, here's what is happening:

    00 00 00 00 00 00 send_keys: 0 398 6
    sending: AT+BLEKEYBOARDCODE=00-00-00-00-00-00-00-00
    ble send: AT+BLEKEYBOARDCODE=00-00-00-00-00-00-00-00
    result: ERROR
    got: ERROR

Any AT+BLEKEYBOARDCODE command that gets sent is returning an error.

If you could shed a light on anything that comes to your mind that can be causing the issue, it would be greatly appreciated.

Thank you!

marhalloweenvt, 2019-09-11:
Hey @fcoury,
Have you look at #6671? Could it be related with this issue? i haven't tried with my pcb yet.

harshitgoel96, 2019-10-04:
@fcoury
Someone got it to work again

https://www.reddit.com/r/MechanicalKeyboards/comments/cpy8bn/photos_her_first_word_was_qqqqqq/?utm_medium=android_app&utm_source=share

fauxpark, 2019-10-10:
I just got one of these Feather BLE boards, and I'm hoping to get some time to investigate this on the weekend.

I've already had a cursory look through the Bluefruit code, and one thing sticks out to me: `send_keyboard()` is not very smart. If NKRO is active, the bytes it sends may well be absolute gibberish, as the NKRO report struct [shares the same space](https://github.com/qmk/qmk_firmware/blob/528ddb79871b76d026c4b2ca8b1debf9feb1e751/tmk_core/common/report.h#L132-L151) as the 6KRO struct. Unfortunately Adafruit's BLE firmware doesn't support NKRO, and they appear to be hitting the same Nordic licensing roadblocks that we are, so there is no source code available.

@harshitgoel96 @fcoury could you try disabling NKRO and see if that does anything?

fcoury, 2019-10-10:
@fauxpark the tests I did so far all had NKRO disabled :(

alan0ford, 2019-10-11:
Sorry for my bad english, if this helps, report my experience with Adafruit feather 32u4
Reference: https://reddit.com/r/MechanicalKeyboards/comments/dd6ozo/handwired_45_bluetooth/?ref=share&ref_source=link
Source file: https://1drv.ms/u/s!AtnBVv-EwzUF5EUC9hvT9pFfjNHv?e=91g7Jp

- every time I turn OFF the PC, I have to redo pairing BT with the keyboard (remove first and re-pairing)
- function KC_PWR don't work (with USB yes)
- function USB_LED_CAPS_LOCK don't work (with USB yes)

yanfali, 2019-10-11:
@alan0ford https://learn.adafruit.com/introducing-the-adafruit-bluefruit-spi-breakout/ble-services because there's no way to know what the adafruit firmware is doing, it's possible they aren't sending these codes correctly. You could try asking adafruit if they have any filters when passing through the hid values.

KC_PWR is 0xA5 this code isn't listed as a USB HID code at all in this list, so it maybe undefined in the adafruit firmware. http://www.freebsddiary.org/APC/usb_hid_usages.php

fauxpark, 2019-10-11:
`KC_PWR` is actually usage page `0x01`, usage `0x81` System Power Down, the keycode value in QMK is just so it can be used as a "basic" keycode. The BLE driver will translate media keys into `BLEHIDCONTROLKEY` AT messages. Adafruit's firmware doesn't support sending arbitrary HID usages, though, so these are the only ones supported: https://learn.adafruit.com/adafruit-feather-32u4-bluefruit-le/ble-services#at-plus-blehidcontrolkey-14-39

alan0ford, 2019-10-12:
@yanfali & @fauxpark very nice, even for a newbie as me, you were clear and comprehensive, thank you!
Apart from than written above, my keyboard works perfectly and I'm very happy...
I take this opportunity to thank you all that worked on this project, dedicating your free time with passion, patience and kindness... thank you again :)

yanfali, 2019-10-12:
@fauxpark looks like we can send raw values as hex. Something to look into

alan0ford, 2019-10-23:
>     * every time I turn OFF the PC, I have to redo pairing BT with the keyboard (remove first and re-pairing)

BT bonding issue solved upgrading feather firmware via app "Adafruit Bluefruit LE Connect"

iamjackchen, 2019-11-21:
> @wez it seems like it's not something on QMK that is causing the issue, here's what is happening:
>
> ```
> 00 00 00 00 00 00 send_keys: 0 398 6
> sending: AT+BLEKEYBOARDCODE=00-00-00-00-00-00-00-00
> ble [...]

hi there, I was wondering how you managed to get the console printouts for debugging - I am working with a Teensy 2.0++ and a BLE SPI friend and I'm currently encountering some issues. Many thanks!

fauxpark, 2019-11-21:
@iamjackchen you'll want to have `CONSOLE_ENABLE = yes` in your rules.mk, the `DEBUG` keycode added to your keymap, and the Teensy connected via USB as well as Bluetooth (the Console feature only works over USB). You can then view the debug messages with the QMK Toolbox.

iamjackchen, 2019-11-22:
> @iamjackchen you'll want to have `CONSOLE_ENABLE = yes` in your rules.mk, the `DEBUG` keycode added to your keymap, and the Teensy connected via USB as well as Bluetooth (the Console feature only works over USB). You can then view the debug messages with the QMK Toolbox.

i see, thanks so much!!

sfcgeorge, 2020-06-14:
I got it working just last week, no problems at all so recent versions are fine. I based my board code off the handwired 42 https://github.com/qmk/qmk_firmware/blob/e258b10d71049645c2e6ce50f652ce3f11578603/keyboards/handwired/42/rules.mk

pcunnane, 2020-12-24:
Anyone managed to get this to work? I downloaded the source that @sfcgeorge linked to but it still didn't work for me. Keyboard works with USB plugged in, but not while on the only powered by the LiPo battery.

iamjackchen, 2021-02-15:
> Anyone managed to get this to work? I downloaded the source that @sfcgeorge linked to but it still didn't work for me. Keyboard works with USB plugged in, but not while on the only powered by the LiPo battery.

I have it working but I built the firmware from scratch (i.e. creating a new keyboard within qmk) instead of modifying an existing keyboard/key map setup. Try that, it'll be easier to ensure hardware-software compatibility; another thing is you might want to have a key mapped to manually switch/override the data transmission (but/usb) mode instead of using the default auto. Make sure NKRO is disabled. Hope all of this helps

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc