qmk/qmk_firmware #5440: How could I rearrange the top and shift layers?

HMarxen, 2019-03-18:
Hi,
So far, I had used just a second layer for shift but that can be problematic when I just want to hold down shift e.g. in games or to select several files. I’ve found this post:
https://github.com/qmk/qmk_firmware/issues/817
So, this bit of code works:

`static bool rereg_shift;
case MA_LESS:
  if (record->event.pressed) {
    if (keyboard_report->mods & MOD_BIT(KC_LSFT)) {
						rereg_shift = true;
						unregister_code (KC_LSFT);
						register_code (DE_ACUT);
					} else {
						rereg_shift = false;
						register_code (DE_LESS);
					}
				} else {
					if (rereg_shift) {
						unregister_code (DE_ACUT);
						register_code (KC_LSFT);
					} else {
						unregister_code (DE_LESS);
					}
				}
				return false; break;`

But it tends to lock shift or run away with <<<<<<<<<<<<<<<´´´´´´´´´´´´´´´´´´´´´´´´´´. 
How could I make this work? These issues are discussed in the mentioned post but I need help to put it into workable code.
Is there perhaps a much more elegant solution built into QMK? Because I’m surprised that there seem to be so few people who try to rearrange the shift layer.

drashna, 2019-03-19:
What are you trying to do specifically?

HMarxen, 2019-03-19:
I want to reasign the characters that are typed when holding shift. For example, instead of "1" becoming "!" when holding shift, I would like it to type "#". The German keyboard is quiet arbitrary in how easy it is to get to more or less often used characters. And we even have to use an extra Alt Graph layer.

drashna, 2019-03-20:
Ah, okay.

Just a heads up, part of this is handled by the OS itself, as the keyboard sends "HID Codes", and not letters.  The OS takes these, and uses the keyboard layout specified to determine what character should be outputted. 

Also, you may want to use `register_code16` and `unregister_code16` here, instead, just in case those keycodes have modifiers applied to them. 

Otherwise, I know there was a recent PR for a keymap that had this sort of functionality, actually. But I'm not sure which.

HMarxen, 2019-03-20:
THX. So I figured, it is probably easiest to keep my extra override layer for shift but just make shift a macro, which presses shift and switches layers at the same time. However, this just ends up with a stuck shift key:

> case MA_LSFT:
				if (record->event.pressed) {
					layer_on(_SH);
					register_code16 (KC_LSFT);
				}
				else{
					layer_off(_SH);
					unregister_code16 (KC_LSFT);
				}
				return false; break;

closed 2019-11-21 by drashna