qmk/qmk_firmware #5470: Modifier Keys release too early

LuckyLukert, 2019-03-23:
## Describe the Bug
In the german layout neo2, caps acts as a modifier and the combination shift-caps is another one. Therefore it would be beneficial to have a key which acts as shift and caps together.
I tried the keycode mykey := LSFT(KC_CAPS)
According to the documentation, the shift should be only released when I release mykey. However, according to xev, shift gets released as soon as I press another key. For example, if I do 

mykey-down, KC_R-down, KC_R-up, mykey-up

then I get:
lshift-down, caps-down;      lshift-up, r-down;      r-up;      caps-up

i.e., the lshift-up is way to early to have effect on the KC_R.

## System Information

 - Keyboard:  custom dactyl-manuform
 - Operating system: ubuntu
 - AVR GCC version: 5.4.0
 - ARM GCC version: 6.3.1
 - QMK Firmware version: newest
 - Any keyboard related software installed? 
   - [no] AutoHotKey
   - [no] Karabiner
   - [no] Other:

MTRNord, 2021-04-24:
I think I run into this or a similiar issue  with the zsa moonlander. I have german backslash which is `alt_gr` plus `ß` in the german layout. But pressing it windows 10 only recognizes `ß`. This seems out of nowhere to happen for all my german keys. Only change I made is update to Windows 10 Build 21364.1 (at this time latest dev version)

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc