qmk/qmk_firmware #5488: Remove build option firmware size impacts from rules.mk

fauxpark, 2019-03-26:
For example:

```make
...
BOOTMAGIC_ENABLE = no    # Virtual DIP switch configuration(+1000)
MOUSEKEY_ENABLE = yes    # Mouse keys(+4700)
EXTRAKEY_ENABLE = yes    # Audio control and System control(+450)
...
```

These numbers are misleading, because they are outdated and as @drashna mentioned here: https://github.com/qmk/qmk_firmware/pull/5485#issuecomment-476420194, dependent on the compiler version. It seems there is not much point having them around since they will simply never be accurate.

But... these are in *every* rules.mk, so I won't bother making a PR for this, even though I strongly feel these kinds of changes should be done sooner rather than later as QMK grows in popularity.

drashna, 2019-03-28:
@jackhumbert @skullydazed 

Personally, I'm in favor of this.  They're not accurate, and I don't think we should keep them in the repo.

noroadsleft, 2019-03-28:
Strongly in favor.

roguepullrequest, 2019-04-27:
Would this z

fauxpark, 2019-04-27:
> Would this z

Probably not.

PR #6947 (2019-10-19): Remove build option firmware size impacts

## Description

Removes references to build option sizes in each keyboard's `rules.mk` (e.g. (+1000)).

Includes the script used to do this in `util/`, `rules_cleanup.sh`.

I have also removed references to these sizes in the documentation, where appropriate.

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [x] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [x] Documentation

## Issues Fixed or Closed by This PR

Fixes #5488 - as @fauxpark comments;
> These numbers are [... cut]
closed 2019-10-19 by drashna