qmk/qmk_firmware #5514: Fixing "Sorry there is no default for the miuni32 keyboard... yet!"

adam-lee, 2019-03-30:
I am sure it's documented somewhere, but I can't seem to find the answer. Where or how can I define the default layout for a keyboard (say, [Miuni32](https://github.com/qmk/qmk_firmware/tree/master/keyboards/miuni32))? When I click on "Load Default" from [QMK Online Configurator](https://config.qmk.fm/#/miuni32/LAYOUT_ortho_3x11), I get the message:

    Sorry there is no default for the miuni32 keyboard... yet!

I'd like to be able to use the online configurator to edit layouts based on a default mapping. Also please note that I can actually download a *.hex* from the [QMK List of Keyboards](https://qmk.fm/keyboards/). Ty.

drashna, 2019-03-30:
Technically, wrong repo then. 

You'd need to add a json file here: https://github.com/qmk/qmk_configurator/tree/master/public/keymaps

zvecr, 2019-10-28:
Fixed by qmk/qmk_configurator#433

closed 2019-10-28 by zvecr