qmk/qmk_firmware #5518: Tune `TAP_CODE_DELAY`

skullydazed, 2019-03-31:
A lot of keycodes require a delay or the host will consider it spurious input and ignore it. We should tune this value to work in most situations. This will require empirical testing on several different boards/os's.

OS's:

* Linux
* macOS
* Windows

Keyboards:

* atmega32u4 based board
* stm32f303 based board
* vusb board

You can help this issue progress by testing any of the above combinations and reporting your results back here. We would like to test 0, 5, 10, 20, and 50 ms delays with the following keycodes:

* `KC_A`
* `KC_VOLU`
* `KC_MS_U`
* `KC_BTN1`
* `KC_HASH`

Based on a conversation in #help it seems that 10 ms was enough for `tap_code16(KC_VOLU);` to work on linux when it previously did not work. Normal keypresses happen in somewhere between 20-90 ms depending on whether you're typing normally or trying to press a key as quickly as possible. On macOS the capslock key needs to be held at least ~200ms or it will not activate, but this is a special case and 200ms is way too long to wait for tapped keys.

aleksclark, 2019-04-01:
So, testing environment:
- Arch Linux
- ProMicro macro pad
- AwesomeWM

I wanted buttons for "move to workspace" and "switch to workspace", which with my current config use `KC_LGUI` + `KC_LSFT` + `KC_1`, and `KC_LGUI`  + `KC_1` respectively.

When setting up this macro, I found that 30ms was the lower bound for a functional macro, but it seems like the issue was with the `KC_LGUI` keycode in particular - I'd get `1` and `!` intermittently, and it would happen more frequently when I had first typed on another keyboard, and then hit the macro pad.

However, this is at least partially an artifact of the WM - `xev` showed the key presses in the correct sequence, but awesomewm ignored the commands when the delay was under 30ms.

drashna, 2019-04-01:
Testing some changes to LT/MT to use tap_code_delay, and no noticeable effect for the rgb animations on the EZ.

XScorpion2, 2019-04-01:
are you delaying inline 10, 20, 30, etc ms? or adding the key to a queue for release in the future?

drashna, 2019-04-10:
using the `tap_code` function, so it's inline delay

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc