qmk/qmk_firmware #5535: I can get stuck in a second layer if I release the first layer.

finkSpell, 2019-04-02:
I can get stuck in a second layer on my Ergodox EZ.

## Describe the Bug
I hold a tap toggle button to activate Layer 2. From Layer 2, I tap a tap toggle button to activate Layer 3. I then let go of the button for Layer 2. I'm stuck in Layer 3. It is possible to get out of this by reactivating Layer 2, but it's confusing. My expectation is that tapping to toggle off Layer 3 either returns me to Layer 2 or the base Layer 0.

The layout I'm using is here:
https://configure.ergodox-ez.com/layouts/ON69/latest/2

1. Press and hold the left thumb button to activate the Arrows 2 layer.
2. Tap to the TT 3 button to activate Layer 3.
3. Let go of the thumb button for TT 2.
4. Tapping the TT 3 button doesn't deactivate layer 3.

You have to turn layer 2 back on before you can get out of layer 3.
## System Information

 - Keyboard:  Ergodox EZ
   - Revision (if applicable):  
 - Operating system: Windows 10
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context
I've never compiled QMK so I don't know about the info being asked for above. The GCC and QMK versions.

drashna, 2019-04-02:
This is normal, actually.  The "TT(3)" keycode is on layer 2.  When you release layer 2, that keycode is no longer on the stack. 

You'd want to manually add it to layer 3, as well.  Well, you'd need to add `TO`, since the Ergodox EZ configurator won't let you add `TT` to a lower layer. 

@ezuk @fdidron 
Not sure there is a good way to handle/fix this, aside from forcibly adding the TT code to the destination layer, unless explicitly defined.

finkSpell, 2019-04-02:
What constitutes a lower layer? I'm confused because I added TT 3 to layer 2, which seems like a lower layer. But I see what you mean: the configurator won't let me add a TT key to layer 3.

drashna, 2019-04-02:
0 is the lowest layer, And each layer "stacks" on top of that.  
https://docs.qmk.fm/#/keymap

But yeah, since you turn off layer 2, the TT(3) keycode is gone. And yeah, you can't add a layer key to a lower level layer, unless it's `TO`.

ezuk, 2019-04-03:
not too much we can do about this from the configurator standpoint imho, unfortunately. Quite hard to validate.

drashna, 2019-04-03:
@ezuk I tested this, and did validate that it happens as described.  
Also, adding TT(3) to layer 3 fixes the issue, but it's a hacky hack

drashna, 2019-04-05:
[image] [Layer Issue](https://trello.com/c/39WoehJh/16-layer-issue)

drashna, 2019-04-05:
@finkSpell  The best option here is to just add a `TO` key on the 3rd layer, so you don't get stuck on it. (though, you can always hold the TT(2) key, as that will still work. 

And I'm closing this for now.  If you think it should be re-opened, then by all means.

closed 2019-04-05 by drashna