qmk/qmk_firmware #5542: Update kinesis stapelberg readme

jlquinn, 2019-04-03:
This readme references a list of parts for building the replacement Kinesis controller.  This part:

2 pin right angle header for reset | 1 | Digi-Key | 3M9467-ND

is obsolete and no longer available.  My searching suggests that
Digi-Key ‎952-2244-ND‎ is an equivalent part.

Also, a similar SparkFun cable could be found at Digi-Key with part number:  PRT-08672-ND

drashna, 2019-04-03:
If you wanted to update the documentation and open a PR, that would be awesome.
(you can just edit it on github, and it should open a PR for you)

If not, somebody can/will.

gooodmorningopenstack, 2019-04-23:
Will take a look ASAP to help a bit :)

noroadsleft, 2019-04-23:
I think #5557 fixes this if completed.

sanyamdogra, 2019-10-08:
Yup #5557 does the work!

closed 2019-10-08 by noroadsleft