qmk/qmk_firmware #5544: Two FN keys holding issue

anjari, 2019-04-03:
It's hard to describe it so I will just show the steps to produce it:

1) Assume you have mapped FN + W/A/S/D as arrow keys. Then you have two FN keys.

2) Hold 1st FN, then press W/A/S/D keys. It will function as arrow keys as expected.

3) While 1st FN still being held, hold the 2nd FN key.

4) Then press the W/A/S/D keys. It still functions as arrow keys.

5) Relese the 1st FN (note that the 2nd FN key is still being held)

6) Then press the W/A/S/D keys. This time the arrow keys don't work, it will produce normal w/a/s/d letters.

Expected result: since the 2nd FN key is still being held, pressing W/A/S/D keys should still function as arrow keys.

I also found this issue in TMK firmware.

FYI I'm using KBParadise V60 Type R polestar edition.

drashna, 2019-04-03:
By FN keys, I'm assuming that you mean layer keys.

If that's the case, then ... unfortunately, this may be normal.   

The code doesn't check to see if the layer is being enabled, but instead enables/disables the layer based on if key is being pressed or not.  

In this case, releasing the first key turns off the layer, regardless of the status of the second key. 

I'm not sure there is a solution here, at least that isn't complex, or involves creating custom macros.

uggnot, 2019-04-03:
I've been thinking about this problem a lot recently and i think the only way to do it would be with custom keycode macros that track layer state rather than using the native layer change mappings.  Then you'd have to put checks on keydown and keyup to see if the layer should be enabled/disabled and save the layer state to the persistent state variable.

drashna, 2019-04-03:
@uggnot Yup, exactly.  Nothing wrong with that, either.

Salt-Peanuts, 2019-04-06:
For this reason, I have two identical function layers, each assigned to its own function layer key (dual-role space and backspace on my Kinesis).

anjari, 2019-04-07:
Thanks for the replies. So would there be code update to get this functionality?

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc