qmk/qmk_firmware #5552: Not clear on how to set custom static RGB gradient

capaldo, 2019-04-04:
I'm attempting to simply set the static gradient to be red, blue, green, yellow, orange and remove the pink/purple/teal/etc.

If anyone can provide some guidance because I've been at this for about 6-7 hours and cant seem which file I need to define the hues that I want to display.

drashna, 2019-04-04:
You mean this?
```c
const uint16_t RGBLED_GRADIENT_RANGES[] PROGMEM = {360, 240, 180, 120, 90};
```

If so, this should be able to go into your `keymap.c` 

However, I don't think you can get rid of the colors you don't want, short of implementing your own code.

That said, try: 
```c
const uint16_t RGBLED_GRADIENT_RANGES[] PROGMEM = {360, 240, 120, 60, 39};
```

capaldo, 2019-04-06:
Did not work unfortunately. Thank you for the help anyways.

drashna, 2019-04-06:
Okay, in that case, you'd need to implement this manually, yourself. 

This is the code that actually creates the effect:
https://github.com/qmk/qmk_firmware/blob/master/quantum/rgblight.c#L553-L567

capaldo, 2019-04-08:
> Okay, in that case, you'd need to implement this manually, yourself.
> 
> This is the code that actually creates the effect:
> https://github.com/qmk/qmk_firmware/blob/master/quantum/rgblight.c#L553-L567

Thank you for your help. Looks like its time to learn some C!

drashna, 2019-04-08:
Well, that should get you started, at least.

Basically, disable animations, and run that block of code in `matrix_scan_user`, and it should get you off and running.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc