qmk/qmk_firmware #5569: Bad Lag sometimes happens with I2C on split boards

nooges, 2019-04-06:
On some of my builds, if USE_I2C is used, that half and the whole board lags a bunch in registering keypresses. 

## Describe the Bug

I’ve tried to trace things down a bit, like erasing EEPROM, compiling with and without EE_HANDS and RGB, but no luck so far. Some of my halves work fine.

## System Information

 - Keyboard:  Levinson Rev 2 & 3
   - Revision (if applicable):  
 - Operating system: Mac
 - AVR GCC version: 8.3

 - ARM GCC version: 

 - QMK Firmware version: Latest

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

drashna, 2019-04-06:
I've actually seen this myself.  
Usually `make clean` and recompiling fixes the issue. Heck, even just recompiling will fix it.  So it's an intermitting issue.

smyjpmu, 2019-06-10:
It is indeed happening on my Iris 2.8 slave side only and recompiling doesn't seem to fix it.

nooges, 2019-06-10:
TODO For self (or anyone trying to fix): Connect I2C lines to oscilloscope and check signal to check that things like rise times look alright.

nooges, 2019-06-19:
Traced issue down to the default I2C clock rate being changed to 400kHz from 100kHz for split_common boards on refactor of split_common to use main I2C driver.

ethanspitz, 2019-07-12:
I found this myself as well. I'm building a handwired, and I didn't yet wire the I2C on the left side (no pull ups) and the typing speed was terrible. Read this, and it all made sense. I put the pullup resistors on I2C and now it's perfectly fine.

nooges, 2020-02-24:
Somewhat related: https://github.com/qmk/qmk_firmware/issues/7522

closed 2020-02-24 by nooges