qmk/qmk_firmware #5590: Fix qmk/ChibiOS submodule to facilitate upgrading

pelrun, 2019-04-09:
Currently qmk/ChibiOS is forked off an old unofficial SVN mirror that is no longer updated. We should make it a fork of the official mirror ChibiOS/ChibiOS instead.

Unfortunately the histories of the two mirrors are completely separate, as they independently generated git commit id's for each SVN commit. This makes the transition a little more difficult.

I propose the following steps be taken:

1. Delete the qmk/ChibiOS repo
2. Fork ChibiOS/ChibiOS into qmk
3. Add https://github.com/mabl/ChibiOS as a remote.
4. Make a new branch that points to the commit id (587968) qmk currently uses from that remote
5. Push it up to github. (this way QMK's current submodule references are still valid)
6. Later when we want to update, we can seamlessly move to the correct commit in the active history.

drashna, 2019-04-09:
ChibiOS-contrib is using the better repo, and can be easily updated (the patch we use is in the main repo, now).

Also, to properly update things, there is some work that will be needed to make sure that QMK Firmware is up to date with ChibiOS changes, so it will compile properly.

GitWellBack, 2019-04-11:
Are we talking about potentially migrating to a newer ChibiOS version as well (since we seem to still be on 17.6, which is no longer supported)?

I ask because I'm about 80% of the way through adding a new MCU to QMK/ChibiOS and I was considering moving the whole underlying OS to ChibiOS 19.1 when I have it successfully integrated this work into current QMK.

pelrun, 2019-04-12:
This is only for fixing a problem with the ChibiOS repo fork we're using, but yes, the hope is that sometime in the future we'll be able to upgrade to a newer version of it. There's a bunch of work involved though.

pelrun, 2019-04-12:
Oh, and if you're porting a new mcu, it should really go on the ChibiOS-contrib repo instead, that's what it's for. The structure is identical, it's handled as an overlay. (And consider submitting it upstream!)

GitWellBack, 2019-04-12:
Oh it'll definitely be submitted when it's fully fleshed out (and it's in contrib, I just wanted to chime in about this topic ;-)) You'd get the 19.1 stuff as well, if I end up going that way later. Good luck with the fix-list (things are always a bunch of work, but we do them anyway!)

closed 2022-06-17 by tzarc