qmk/qmk_firmware #5595: Combo feature intermittently skips keystrokes

noahfrederick, 2019-04-10:
## Describe the Bug

The combo feature (`COMBO_ENABLE`) can result in certain keys not registering while typing, after https://github.com/qmk/qmk_firmware/pull/2561.

## System Information

 - Keyboard:  Preonic
   - Revision (if applicable):  Rev 3
 - Operating system: macOS
 - AVR GCC version: 7.3.0
 - ARM GCC version: 7.2.1 20170904 (release)
 - QMK Firmware version: 0.6.328
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [x] Karabiner
   - [ ] Other:

## Additional Context

```c
// config.h
#define COMBO_TERM 50
```

```c
// user.c
enum combo_events {
  MY_COMBO_AND,
  // ...
};

const uint16_t PROGMEM combo_and[] = { KC_A, KC_N, COMBO_END };
// ...

combo_t key_combos[COMBO_COUNT] = {
  [MY_COMBO_AND] = COMBO_ACTION(combo_and),
  // ...
};

void process_combo_event(uint8_t combo_index, bool pressed) {
  switch(combo_index) {
    case MY_COMBO_AND:
      if (pressed) {
        SEND_STRING("and ");
      }
      break;
    // ...
  }
}
```

_Typing_ `want` sometimes outputs `wnt` (expected `want`), and _chording_ `an` sometimes outputs `nd` (expected `and`).

rsheldiii, 2019-04-10:
if `SEND_STRING` isn't working this sounds like a deeper issue, unless maybe an erroneous false from [`process_combo`](https://github.com/rsheldiii/qmk_firmware/blob/master/quantum/quantum.c#L274) would interfere with `SEND_STRING`? 

if it occurs intermittently, does lengthening `COMBO_TERM` make the bug occur less often, or more often? does deliberately holding the combo longer (mostly the same approach) do anything?

noahfrederick, 2019-04-10:
@rsheldiii Thank you so much for your work on this feature.

> if it occurs intermittently, does lengthening `COMBO_TERM` make the bug occur less often, or more often?

In my testing, lengthening `COMBO_TERM` lessened (or even entirely eliminated) the issue (which is not viable for daily use because I get `wand t` for `want`, i.e., the combo triggers during sequential typing).

> does deliberately holding the combo longer (mostly the same approach) do anything?

Interesting—Yes, holding the combo longer (until the entire string is printed, for example) doesn't exhibit the dropped keys.

Incidentally, tapping the combo as quickly as possible sometimes results in a stuck key:

Chord `an` - > `nnnnnnnnnnnn` (repeating until tapping `n` key again)

I can open a separate issue for that if you don't think it's relevant.

rsheldiii, 2019-04-11:
I can't rule out repeating characters being caused by this bug, that's definitely been an issue with combos before but maybe it's more frequent now. It sounds like this is an issue with process_combo incorrectly sending a return value, swallowing a letter it shouldn't be swallowing; looking at [SEND_STRING's](https://github.com/qmk/qmk_firmware/blob/91b18e263f6373994613c290129e49f18e17cef1/quantum/quantum.c#L907) code it literally registers and unregisters keypresses, so I think that's still subject to `process_record_quantum `. I might have botched something during the merge conflict resolution or maybe there's something I'm not seeing. I'll see if I can replicate this when I get home today.

PR #5610 (2019-04-14): fix combo enabling logic

## Description

I botched the merge conflict resolution in #2561 , inverting the logic for when combos should be re-enabled. This PR fixes #5595 that came as a result; on master with combos I can reproduce stuck keys and swallowed keypresses when using `SEND_KEYS()` but with these changes I cannot. It'd be great if @noahfrederick could confirm!

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Close [... cut]
closed 2019-04-14 by drashna