qmk/qmk_firmware #5597: Is it possible to use FastLED lib8tion library in custom animations?

alecgeatches, 2019-04-10:
I'm very rusty in C, so I apologize if this is a C-ignorance issue rather than something QMK-specific.

The [FastLED lib8tion library](https://github.com/qmk/qmk_firmware/pull/5372/files#diff-9d68415829cd94539984782c0ced9552R6) was recently added in #5372 by @XScorpion2. There's a ton of great functionality in that library (scaling, fast `sin8`, easing functions) that I would like to use in custom animations called from my `rgb_matrix_indicators_user`.

I'm having an issue using any of the libation functions in my custom code. If I try adding `#include "lib/lib8tion/lib8tion.h"` to my keymap file, I get a linker error:

```bash
$ make ergodox_ez:alecg
QMK Firmware 0.6.328
Making ergodox_ez with keymap alecg

# ...

 [OK]
Linking: .build/ergodox_ez_alecg.elf                                                                [ERRORS]
 |
 | .build/obj_ergodox_ez_alecg/quantum/rgb_matrix.o:(.rodata.b_m16_interleave+0x0): multiple definition of `b_m16_interleave'
 | .build/obj_ergodox_ez_alecg/keyboards/ergodox_ez/keymaps/alecg/keymap.o:(.rodata.b_m16_interleave+0x0): first defined here
 | collect2: error: ld returned 1 exit status
 |
tmk_core/rules.mk:290: recipe for target '.build/ergodox_ez_alecg.elf' failed
make[1]: *** [.build/ergodox_ez_alecg.elf] Error 1
Make finished with errors
Makefile:535: recipe for target 'ergodox_ez:alecg' failed
make: *** [ergodox_ez:alecg] Error 1
```

From what I understand, this is due to the `b_m16_interleave` constant being defined in the header file *lib/lib8tion/trig8.h* included from *lib8tion.h*. Since the constant is defined in the header file, including the same header from anywhere else causes a linker error becuase it's already included in *quantum/rgb_matrix.c*.

Is there any way to make use of the library from non-rgb_matrix code? Thank you!

pelrun, 2019-04-10:
Simplest fix is to add 'static' to the definition of `b_m16_interleave`, which will remove any external visibility. A more complete fix will be to move the definition to a C file and include it in the build, but it's not necessary unless you're desperate to save a dozen bytes of ram or flash.

alecgeatches, 2019-04-10:
@pelrun Thank you for the information! I made a PR with your second approach in #5598.

XScorpion2, 2019-05-13:
This has been merged and can be closed

closed 2019-05-14 by drashna