qmk/qmk_firmware #5620: Eager PR issue

drashna, 2019-04-15:
## Describe the Bug

Eager Per Row debouncing causes system to wake up from sleep.

This only happens with Eager Per Row debouncing, and ONLY on the Ergodox EZ, from what I can tell.

Tested with Sym_G and Eager_PK, and no issue with sleep. Tried Eager_PR on the Iris, no issue. 
Reverted #5498 also fixes the issue.  So there is something here that is causing the Ergodox EZ to stay awake. 

## System Information

 - Keyboard:  Ergodox EZ
 - Operating system: Windows 10 x64
 - AVR GCC version: 8.3.0
 - QMK Firmware version: 0.6.334
 - Any keyboard related software installed?  No

alex-ong, 2019-04-15:
Looking at `master`:
```
void debounce(matrix_row_t raw[], matrix_row_t cooked[], uint8_t num_rows, bool changed)
{
  uint8_t current_time = timer_read() % MAX_DEBOUNCE;
  update_debounce_counters(num_rows, current_time);
  transfer_matrix_values(raw, cooked, num_rows, current_time);
}
```
looks like (bool changed) isn't being used.
I swear i had a pr that fixed this, let me go find it...

drashna, 2019-04-15:
Just to clarify, it's the PC.  I put it to sleep, and it wakes up immediately.  If i revert the commits for the debouncing change, or use a different debounce type, it sleeps just fine.

drashna, 2019-04-15:
As for why it may be causing issues, my best guess would be some interaction with the IO Expander, since this doesn't happen with my other boards. And I tried it with the lights disabled too

alex-ong, 2019-04-15:
> As for why it may be causing issues, my best guess would be some interaction with the IO Expander, since this doesn't happen with my other boards. And I tried it with the lights disabled too

Right - but changing debounce method shouldn't change interaction with IO-Expander (apart from the polling rate changing; faster debouncing will make IO-Expander be called more)

PR #5621 (2019-04-15): Eager pk/pr optimization

Eager pk/pr will now have effectively 0 runtime when no debouncing is required

## Description
A boolean stores whether we are currently debouncing anything.
We can skip scanning the matrix if we are are not debouncing.

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Closes #5620 

## Checklist

- [x] My code follows the code style of this project.
- [ ] My change requires a c [... cut]
closed 2019-04-15 by drashna