qmk/qmk_firmware #5631: All ARM Keyboards do not work on Ubuntu 18.04

awkannan, 2019-04-16:
On Ubuntu 18.04, ARM powered keyboards are not recognized by default.

The OS sees the device, but no input shows up until you run hid_listen, eventually hid_listen picks up the device and then you can start typing.

The keyboard will intermittently stop working, and you will have to run hid_listen again to get it to work.

@ mxgian reproduced this issue with a Planck rev 6, Instant60, and Ortho60 board. So it's not just limited to f303, but is also present on f103 and f072.

mxgian, 2019-04-16:
hid_listen only opens the raw device to read input.  i can also 'make the keyboard' work by cat'ting the raw device, in my case 'cat /dev/hidraw0' to start reading the raw device and then the keyboard starts working.  

at least for me the keyboard continues to work until disconnect, but others have reported it will stop working after a few minutes.

yanfali, 2019-04-16:
Looks like Xorg is crashing and taking down the system with it

https://gist.github.com/yanfali/a16bd5a7da633830962d127e81999caa

```
Apr 16 12:00:26 ptah systemd-coredump[1115]: Process 1056 (Xorg) of user 0 dumped core.
                                             
                                             Stack trace of thread 1056:
                                             #0  0x00007f1412572d7f raise (libc.so.6)
                                             #1  0x00007f141255d672 abort (libc.so.6)
                                             #2  0x0000563104dcab5a OsAbort (Xorg)
                                             #3  0x0000563104dc15cf FatalError (Xorg)
                                             #4  0x0000563104dcf3ee n/a (Xorg)
                                             #5  0x00007f1412572e00 __restore_rt (libc.so.6)
                                             #6  0x00007f14125f17c4 __wcslen_sse2 (libc.so.6)
                                             #7  0x00007f14125e0883 wcsrtombs (libc.so.6)
                                             #8  0x00007f141258de34 vfprintf (libc.so.6)
                                             #9  0x00007f1412645c5f __vsnprintf_chk (libc.so.6)
                                             #10 0x0000563104dbc901 Xvscnprintf (Xorg)
                                             #11 0x0000563104dc290f LogVMessageVerb (Xorg)
                                             #12 0x00007f140fe30044 n/a (libinput.so.10)
                                             #13 0x00007f140fe381ad n/a (libinput.so.10)
                                             #14 0x00007f140fe3a6d1 n/a (libinput.so.10)
                                             #15 0x00007f140fe1e944 n/a (libinput.so.10)
                                             #16 0x00007f140fe1eb29 n/a (libinput.so.10)
                                             #17 0x00007f140fe1ec87 libinput_path_add_device (libinput.so.10)
                                             #18 0x00007f140ffd20ff n/a (libinput_drv.so)
                                             #19 0x0000563104da2c27 n/a (Xorg)
                                             #20 0x0000563104d680c2 n/a (Xorg)
                                             #21 0x0000563104d689a3 config_init (Xorg)
                                             #22 0x0000563104daf137 InitInput (Xorg)
                                             #23 0x0000563104d55734 n/a (Xorg)
                                             #24 0x00007f141255f223 __libc_start_main (libc.so.6)
                                             #25 0x0000563104d5630e _start (Xorg)
                                             
                                             Stack trace of thread 1072:
                                             #0  0x00007f1411ba2afc pthread_cond_wait@@GLIBC_2.3.2 (libpthread.so.0)
                                             #1  0x00007f140f3bc254 n/a (i965_dri.so)
                                             #2  0x00007f140f3bbf78 n/a (i965_dri.so)
                                             #3  0x00007f1411b9ca9d start_thread (libpthread.so.0)
                                             #4  0x00007f1412636b23 __clone (libc.so.6)
                                             
                                             Stack trace of thread 1068:
                                             #0  0x00007f1411ba2afc pthread_cond_wait@@GLIBC_2.3.2 (libpthread.so.0)
                                             #1  0x00007f140f3bc254 n/a (i965_dri.so)
                                             #2  0x00007f140f3bbf78 n/a (i965_dri.so)
                                             #3  0x00007f1411b9ca9d start_thread (libpthread.so.0)
                                             #4  0x00007f1412636b23 __clone (libc.so.6)
Apr 16 12:00:26 ptah systemd[1]: systemd-coredump@0-1114-0.service: Succeeded.
```

I haven't done a headless test to see if this affects terminal.

yanfali, 2019-04-16:
This was with a 5.0.7 kernel on Arch Linux, so it's pretty new. Xorg was 1.20.4 so fairly new.

yanfali, 2019-04-16:
https://bugs.freedesktop.org/show_bug.cgi?id=110453 issue filed with Xorg. Not sure what'll happen.

mxgian, 2019-04-16:
thanks for the quick turnaround.  if you need to me generate a dump from my ubuntu system i can do that, just let me know how to generate the coredump.

yanfali, 2019-04-16:
It should be in the journald. I found it by using root and typing `journalctl > afile.txt` and then working backwards in the file. You could look for a REBOOT event as where to start searching.

TellNoLies, 2019-04-16:
I have the same issue as mxgian on Gentoo Linux. My system:
Operating System: Gentoo Linux 
KDE Plasma Version: 5.14.5
Qt Version: 5.11.3
KDE Frameworks Version: 5.54.0
Kernel Version: 4.20.10-gentoo
OS Type: 64-bit
Processors: 12 × Intel® Core™ i7-5820K CPU @ 3.30GHz
Memory: 15.6 GiB of RAM
Installed versions:
cross-arm-unknown-linux-gnueabi/gcc 7.3.0-r3, 8.2.0-r6, 8.3.0-r1
cross-avr/gcc 4.9.4, 7.3.0-r3
x11-base/xorg-drivers 1.20
cross-avr/avr-libc 2.0.0
dev-embedded/avrdude 6.3
sys-devel/crossdev 20190311

Hopefully that is enough information.

yanfali, 2019-04-17:
@TellNoLies feel free to chime in on the bug I filed.

TellNoLies, 2019-04-17:
@yanfali Should I paste the same info on the Bugzilla?

yanfali, 2019-04-17:
@TellnlNoLies I think that would be helpful to have more than one user chime in, especially on different distros

TellNoLies, 2019-04-17:
@yanfali how was that comment that I just left on the bug report? Will that info be helpful?

yanfali, 2019-04-17:
Very nice

TellNoLies, 2019-04-17:
Okay, thanks.

drashna, 2019-04-18:
I just booted up my laptop to test this.   
Ubuntu 18.10 x64 
Kernel is 4.18.0-17-generic.

TellNoLies, 2019-04-18:
@drashna did it do the same for you on Ubuntu?

drashna, 2019-04-18:
> @drashna did it do the same for you on Ubuntu?

I'm assuming you mean 18.04, specifically.  I don't know. I've been using 18.10 for a while, since the ARM GCC binaries are broken on 18.04.   However, I will swap drives out and test this in the next day or so.

drashna, 2019-04-19:
Installed 18.04.  Can't reproduce.  had system on for an hour, testing periodically.

sfines, 2019-04-20:
To date I've reproduced it on 18.04, 18.10, and 19.04, as well as on 3 different machines by 3 different manufacturers.

drashna, 2019-04-26:
Well, if you have mousekeys enabled, it does add a USB HID Mouse device, as well.

sfines, 2019-04-26:
I wonder if that's the problem. I'll disable mousekeys (it's part of the cannonkeys ortho60 default image, if it is in fact on) and see what happens.

airyimbin, 2019-05-28:
For me the keyboard is only recognized if I boot with it plugged in from the start.
If I plug in the keyboard after Ubuntu boots it does not recognize it.
Also I if enable or disable LED it no longer recognizes it and I have to reboot.

I'm on Ubuntu 18.04.

rycwo, 2019-06-24:
@sfines, having the exact same issue on Arch Linux/Centos 6.5 with the Ortho60. Did disabling mouse keys fix the issue for you?

With limited knowledge it looks like the following for me:

- board is recognized;
- bootloader loads the QMK firmware;
- then I get `usb_submit_urb(ctrl) failed: -1`

The board used by the Ortho60 is the [STM32F103C8](https://wiki.stm32duino.com/index.php?title=Blue_Pill).

yanfali, 2019-06-24:
We're not going any response for xorg, you might want to go nag them as it's an xinput issue and it's not clear how to resolve it.

rycwo, 2019-06-24:
@yanfali - X doesn't crash for me, the keyboard just isn't responding as @sfines mentions. It isn't responsive in plain tty for me.

nyanshak, 2019-06-24:
@rycwo I tried disabling mouse keys but that didn't seem to help. I'm less-familiar with debugging this sort of issue unfortunately. I also don't have X crashing; the keyboard isn't responding.

mxgian, 2019-06-24:
unfortunately i dont have any suggestions to get the X-side issues fixed, but a few of us have been successful running hid_listen in the background, this sort of 'jumpstarts' the device and it starts responding, not a solution and a pain if it dies and its the only keyboard you have plugged in, but it's at least a workaround.

airyimbin, 2019-06-24:
I specifically have the Alice PCB from Project Keyboard, and found that hitting the reset button until it reconnects works for me. I have Ubuntu at my job that I don't have sudo access to so hid_listen isn't something I can do. Thankfully within 10-15 presses of the reset button usually connects me back.

rycwo, 2019-06-24:
> I have Ubuntu at my job that I don't have sudo access to so hid_listen 
> isn't
> something I can do.

Same situation for me.

> Thankfully within 10-15 presses of the reset button usually connects
> me back.

Just to be clear... is this 10-15 presses with pauses in between?
Or is it spamming 10-15 presses? Definitely have to give this a try
myself!

airyimbin, 2019-06-24:
So my pcb usually takes 2ish seconds to boot. So press and wait 2 seconds. I have LEDs to help me know when the keyboard connected.

rycwo, 2019-06-24:
Just to keep this thread up-to-date, the reset spamming didn't work for
me, but the hid_listen script running in the background seems to do the
trick.

I would be more than happy to come-up with a patch to fix this issue if
any of the QMK maintainers could give some guidance!

yanfali, 2019-06-24:
Well you could try and do a debug build for xinput and then capture a coredump and see if you can find out where it's crashing.

rycwo, 2019-06-24:
On Mon Jun 24, 2019 at 3:04 PM Yan-Fa Li wrote:
> Well you could try and do a debug build for xinput and then capture a coredump and see if you can find out where it's crashing.

As mentioned by multiple comments. There are no issues with X for some
of the boards, that seems to only be happening for you. As far as it
looks most people have the issue where the board is simply not
responsive. At least until they run hid_listen or pull from the
correct fs /dev/hidraw<n>.

yanfali, 2019-06-25:
I think it's all related. There are no issues if you just run it from a tty. Only from X.

evamvid, 2019-09-09:
I've had this issue as well, but the keyboard behaves exactly the same in a tty as it does in x. In both cases, no keypresses are registered until I run hid_listen as root, at which point it starts working and continues even after I kill hid_listen.

I'm using a BluePill-based Ortho48 from Cannonkeys on Arch Linux (technically AntergOS, which is now dead, but it's Arch for all intents and purposes).

yanfali, 2019-09-11:
So I've been trying to figure out what's going on. I have 1 linux box that crashes, it has Sandy Bridge hardware and it's a laptop. I have one desktop running a server based chipset and it does not crash and it's running exactly the same version of xorg 1.20.5. Still haven't figured out what is the difference. Can people chime in on what hardware they running linux on? Specifically intel chipset would be interesting to know.

yanfali, 2019-09-11:
OK folks, I just confirmed that after linux kernel 5.2.9 the problem is gone. I don't know what they fixed but it was still broken around 5.2.5, so in between those two releases something got fixed in the kernel. It's working for me now. I think it's just a kernel issue.

yanfali, 2019-09-24:
https://github.com/qmk/qmk_firmware/pull/6801#pullrequestreview-292187774 please test this patch if you having Linux issues

krnhotwings, 2019-10-02:
Tested the patch, and it does _not_ fix the linux issue. The patch actually would have made no difference with my config settings (mouse buttons disabled, raw hid disabled, audio disabled, console enabled.) With only console enabled, the outcome of HID ordering is the same before and after the patch is applied.

I'm actually fairly confident that the problem has to do with the `CONSOLE_ENABLE` feature itself. If the `rules.mk` file has the feature turned off, the keyboard will work on linux.

I've also found that, at least on my system, if I have the console enabled, the keyboard will work on linux if I treat it like a PS/2 keyboard: keep the keyboard plugged in on boot. Once replugged or restarted, the keyboard will become unresponsive until next boot. Sometimes the keyboard will get into a funky "frozen" state during bootup, which would require me to restart the keyboard and then restart my system, but the instances that that had happened were pretty rare (meaning, I couldn't replicate the freezing behavior.)

iirc, my home desktop is running Pop OS 19.04 with kernel 5.0.0

yanfali, 2019-10-02:
@krnhotwings do you have any QMK keyboards that are AVR based with console turned on? Would like an a/b comparison. It looks like something chibios is doing is causing linux to puke for console output.

krnhotwings, 2019-10-02:
Yup, I have a Tada68 (atmega32u4). Console enabled, keyboard works fine. Keyboard still works when replugged.

I added debug print statements to `process_record_user` and hid_listen prints as expected.

sidcarter, 2019-11-21:
thanks for that @krnhotwings - disabling console also fixed a similar issue with my alice pcb

ridingqwerty, 2019-12-31:
Just want to add my experience to this issue.

I had been using a Proton C on Ubuntu 18.04 (kernel 4.15.0) for some time now with no issues.  CONSOLE_ENABLE had been set to yes in rules.mk at least as far back as Oct 31 of this year.  

Soon after I added the uprintf statement to process_record_user from [testing and debugging docs](https://beta.docs.qmk.fm/newbs/newbs_testing_debugging) a few days ago, I noticed the keyboard became intermittently spammy or unresponsive unless hid_listen was running as root.

As soon as I commented out the uprintf, but with CONSOLE_ENABLE still set to yes, the issue went away.

stapelberg, 2021-04-03:
I just ran into this issue myself.

The culprit is `CONSOLE_ENABLE = yes` in combination with anything that produces enough debug messages to the console, e.g.  `#define DEBUG_MATRIX_SCAN_RATE` in `config.h` which produces one message every second.

The root cause is that the QMK ChibiOS USB driver does not drain the debug message queue until the hidraw device is opened, e.g. by running `hid_listen`.

You can clearly see this when attaching the `gdb` debugger and setting a watchpoint on `ch.mainthread.state`. Eventually, your main thread will transition into `state=4` (`CH_STATE_QUEUED`, i.e. thread suspended on a queue operation) and never run again, which locks up your keyboard:
```
Program received signal SIGTRAP, Trace/breakpoint trap.
chSchGoSleepS (newstate=4 '\004') at lib/chibios/os/rt/src/chschd.c:300
(gdb) bt
#0  chSchGoSleepS (newstate=4 '\004') at lib/chibios/os/rt/src/chschd.c:300
#1  0x6000a35e in chSchGoSleepTimeoutS (newstate=4 '\004', timeout=4294967295) at lib/chibios/os/rt/src/chschd.c:389
#2  0x6000a7d8 in chThdEnqueueTimeoutS (tqp=0x20000d18 <drivers+268>, timeout=4294967295) at lib/chibios/os/rt/src/chthreads.c:872
#3  0x6000ac00 in osalThreadEnqueueTimeoutS (tqp=0x20000d18 <drivers+268>, timeout=4294967295) at ./lib/chibios/os/hal/osal/rt-nil/osal.h:831
#4  0x6000b112 in obqGetEmptyBufferTimeoutS (obqp=0x20000d18 <drivers+268>, timeout=4294967295) at lib/chibios/os/hal/src/hal_buffers.c:586
#5  0x6000b244 in obqWriteTimeout (obqp=0x20000d18 <drivers+268>, bp=0x20000aef "f4041\267\201", n=1, timeout=4294967295) at lib/chibios/os/hal/src/hal_buffers.c:737
#6  0x6000957c in _write (ip=0x20000cd4 <drivers+200>, bp=0x20000aef "f4041\267\201", n=1) at tmk_core/protocol/chibios/usb_driver.c:83
#7  0x6000902c in sendchar (c=102 'f') at tmk_core/protocol/chibios/usb_main.c:935
#8  0x600081b6 in _putchar (character=102 'f') at tmk_core/common/printf.c:27
#9  0x60007592 in _out_char (character=102 'f', buffer=0x20000b94, idx=12, maxlen=4294967295) at lib/printf/printf.c:153
#10 0x600079b8 in _vsnprintf (out=0x60007571 <_out_char>, buffer=0x20000b94 "\f\030", maxlen=4294967295, format=0x600146ac "frequency: %lu\n", va=...) at lib/printf/printf.c:592
#11 0x6000815c in printf_ (format=0x600146a0 "matrix scan frequency: %lu\n") at lib/printf/printf.c:867
#12 0x6000446c in matrix_scan_perf_task () at tmk_core/common/keyboard.c:121
#13 0x6000470c in keyboard_task () at tmk_core/common/keyboard.c:408
#14 0x600091a2 in main () at tmk_core/protocol/chibios/main.c:303
```

Now, what’s the most elegant way to fix this? I don’t know yet :) Looking at this now.

closed 2021-04-10 by zvecr

purdeaandrei, 2021-04-13:
Hello everyone, please review my improved fix for this issue here:
https://github.com/qmk/qmk_firmware/pull/12576
This current implementation has some issues which are described in my PR/commit messages.

PR #12472 (2021-04-10): ChibiOS USB driver: prevent deadlock with CONSOLE_ENABLE = yes
## Description

Before this commit, attaching an ARM-based (i.e. ChibiOS-based) keyboard that
uses CONSOLE_ENABLE = yes and produces debug messages would deadlock the
keyboard unless one was running hid_listen.

With this commit, dead-locking writes to the queue are detected and prevented.

fixes #5631

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by T [... cut]
commit 7d953332 (2021-04-10): ChibiOS USB driver: prevent deadlock with CONSOLE_ENABLE = yes (#12472)

Before this commit, attaching an ARM-based (i.e. ChibiOS-based) keyboard that
uses CONSOLE_ENABLE = yes and produces debug messages would deadlock the
keyboard unless one was running hid_listen.

With this commit, dead-locking writes to the queue are detected and prevented.

fixes #5631