qmk/qmk_firmware #5660: Copyrighted songs

drashna, 2019-04-20:
Since this has come up recently, while I know we don't want to host copyrighted songs.... it would still be nice to have a way to allow users to use copyrighted songs for their keyboards, that are not in the main repo.  Or via some way that can be handled via a proper/compliant method.

alex-ong, 2019-04-23:
maybe create a new repo called qmk_songs, with song_list.h and a folder full of songs?

drashna, 2019-04-25:
That's one option.  The issue is how to handle stuff. 

Eg, takedown notices, etc.

alex-ong, 2019-04-25:
Well if its in the other repo, that repo will be in charge of dealing with copyrighted songs and takedowns, and qmk would be un-sullied :) Hopefully...

iaman, 2019-05-13:
@drashna If the issue is that there needs to be a way for users to keep songs in this repo locally while preventing QMK from having to deal with them potentially being submoduled into this repo or otherwise hosted in this repo, would a `.gitignore` update to ignore all files of `[list of extensions for music files]` be a good first step, or would that just create issues in terms of trying to put songs with permissive licenses into this repo? (is that even a plan?)

drashna, 2019-05-14:
That's the thing, I don't know the best way to handle this.  And it's partially a legal issue, too. 

I'm not sure the best way to handle this, to be honest.

drashna, 2019-06-17:
Songs have been removed.

closed 2019-06-17 by drashna