qmk/qmk_firmware #5690: Mouse keys makes it (almost) impossible to scroll just one step

Markus00000, 2019-04-23:
## Describe the Bug

When initiating scrolling with mouse keys, two scroll events are sent almost immediately while ignoring the configured interval.

To reproduce, configure a constant scroll speed that is slow enough to scroll single steps. For example:

    #define MK_3_SPEED
    #define MK_W_OFFSET_0       1
    #define MK_W_INTERVAL_0     360

Hover your pointer over one of the input fields and try to scroll one step:

https://css-tricks.com/examples/MousewheelInputs/

The first two events ignore the 360ms interval and are sent right after each other. Subsequent events are spaced 360ms apart.

## System Information

 - QMK Firmware version: 0.6.340

## Additional Context

nathanvercaemert, 2020-04-24:
Is this a discussion that the scroll speed is too fast, or only the single issue that you can't  scroll one step (but after scrolling the initial two steps you can slowly scroll?)

Markus00000, 2020-04-25:
This issue is about not being able to scroll just one initial step regardless of scroll speed.

dhong44, 2020-04-30:
I've been trying to fix this for myself recently and saw that this was an issue and believe I have a pretty good understanding of what the problem is now. I think this is a duplicate of both #1434 and #3370 which were closed. I was playing with a new keyboard layout today and can confirm this is still a problem. The settings for mouse keys can be found in [1] and the intended way to allow single steps is to modify the MOUSE_WHEEL_DELAY which will leave a delay in ms between the first scroll step and the next. 

When this issue was opened this was not working because the mouse wheel settings were not used anywhere in the code, instead the mouse movement settings for delay and interval were also used for wheel delay and interval. This was a problem since mouse movement generally has much shorter delays and intervals than the wheel. This was fixed by #6685 when the settings for movement and wheel were decoupled. The problem is now, there are 2 variables for time when mouse movement was last changed and when mouse wheel was last changed which are set based on `last_time` which is updated after `mousekey_send()`. `mousekey_send()` is called the first time a mouse moving key is pressed and doesn't update `last_time_w` which is why the first time scrolling goes an extra step.

https://github.com/qmk/qmk_firmware/blob/faaff8039967ea771fa357f315f085ad3c961f6c/tmk_core/common/mousekey.c#L128-L147

https://github.com/qmk/qmk_firmware/blob/faaff8039967ea771fa357f315f085ad3c961f6c/tmk_core/common/action.c#L364-L379

I think the mousekey area still needs some changes since there should be separate variables for acceleration and number of repeats for both movement and wheel. The work in #7335 looks like it would fix this issue and a lot of the other problems in this area but overlap a lot with #6685. If you need a working hacky solution right away you could try this for now. https://github.com/dhong44/qmk_firmware/commit/f8eee7b90493a1910c93163f4439a0006d3b366d

[1] https://github.com/qmk/qmk_firmware/blob/master/docs/feature_mouse_keys.md

Silent-Hunter, 2020-05-22:
This is a very annoying issue for me, as I cannot use the mouse wheel keys in any games because it zooms through weapon or hotbar selections. Any chance of this being fixed? I have a Planck EZ.

dhong44, 2020-05-23:
I'll open a pr to fix this. I'm a new contributor and didn't want to take it if anyone else was working on it. Since it looks like no one else is I'll try to do it.

nathanvercaemert, 2020-07-17:
As far as scroll speed is concerned, I've found that the mouse wheel interval allows me to control
the speed of the scroll well enough for my liking. It is worth noting that I started tweaking this value only after I used my computer's system settings to turn down the scroll speed as low as it will go.

```
#define MK_W_INTERVAL_0     50 /* Tweak this value until you find what suits you. */
```

I approached the issue of not being able to scroll a single step using a different approach. I defined some custom keycodes for one-step scrolling, and I use them instead of the mouse wheel keys when I only want to move one step. I actually have an MO layer that I throw over the mouse wheel keys that lets me click to scroll one step at a time. Here are the keycodes and the corresponding functionality:

```
enum custom_keycodes {
  RGB_SLD = EZ_SAFE_RANGE, /* Note that the EZ comes with this keycode. */
  MS_WH_UP,
  MS_WH_DOWN,
  MS_WH_RIGHT,
  MS_WH_LEFT,
};
```

```
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
    case MS_WH_DOWN:
        if (record->event.pressed) {
            SEND_STRING(SS_TAP(X_MS_WH_DOWN));
        }
        break;
    case MS_WH_UP:
        if (record->event.pressed) {
            SEND_STRING(SS_TAP(X_MS_WH_UP));
        }
        break;
    case MS_WH_LEFT:
        if (record->event.pressed) {
            SEND_STRING(SS_TAP(X_MS_WH_LEFT));
        }
        break;
    case MS_WH_RIGHT:
        if (record->event.pressed) {
            SEND_STRING(SS_TAP(X_MS_WH_RIGHT));
        }
        break;
    }
    return true;
}
```

dhong44, 2020-09-18:
The custom keycodes shouldn't be necessary now that #9174 is merged. I think this issue should be closed now.

Markus00000, 2020-09-18:
#9174 resolved this issue. I can now reliably scroll just one step. Thank you, @dhong44!

closed 2020-09-18 by Markus00000