qmk/qmk_firmware #5691: Momentary mode of mouse keys resets speed when it should not

Markus00000, 2019-04-23:
## Describe the Bug

Momentary mode is meant to keep a certain speed while the respective key is pressed. When switching speeds, users might concurrently press two speed keys for a moment. This resets the speed even if a speed key is still being pressed.

### How to reproduce

First, enable constant speeds and momentary mode:

    #define MK_3_SPEED
    #define MK_MOMENTARY_ACCEL

1. Move pointer while holding any speed key (e.g. `KC_ACL0`). The pointer should move at the set speed (0).
2. Concurrently press another speed key (e.g. `KC_ACL1`). The pointer should change its speed accordingly (1).
3. Keep holding the second speed key (e.g. `KC_ACL1`) while releasing the first speed key (e.g. `KC_ACL0`). The pointer speed is reset to the unmodified speed.

At this point, the pointer should still move at the speed of the second speed key (e.g. `KC_ACL1`) which is still being pressed.

## System Information

 - QMK Firmware version: 0.6.340

NorthGuard, 2021-09-09:
I have this problem as well. 
Is this resolved or do people just have different preferences?

Markus00000, 2021-09-10:
> Is this resolved or do people just have different preferences?

It is not resolved. Also, I would argue that pointer speed should always respect pressed speed modifier keys. Having the pointer move at default speed even while, let’s say, `KC_ACL1` is pressed leaves no obvious way to switch to `KC_ACL1` speed. One has to release the modifier key and press it again. This goes counter to the way other modifiers (like Shift) work.

tzarc, 2022-06-18:
Closing due to inactivity.

Markus00000, 2022-06-18:
This issue still persists. The bot message said it will be closed “unless it is tagged”. It is already tagged *discussion.*  I would rather tag it as *bug.*

tzarc, 2022-06-18:
> This issue still persists. The bot message said it will be closed “unless it is tagged”. It is already tagged *discussion.*  I would rather tag it as *bug.*

Done. Would be great if you could dig into the code and find out how to fix it. This sitting here for a couple of years with nobody tackling it makes the overall signal-to-noise ratio worse off -- everyone else has seemingly had higher priority things to address instead.

arutonee1, 2025-03-09:
@Markus00000 @NorthGuard @tzarc Not sure if any of you are still interested in a fix for this, but I found one after 10 minutes of digging around the qmk repo.

```diff
--- a/quantum/mousekey.c
+++ b/quantum/mousekey.c
@@ -618,11 +618,11 @@ void mousekey_off(uint8_t code) {
     else if (IS_MOUSEKEY_BUTTON(code))
         mouse_report.buttons &= ~(1 << (code - QK_MOUSE_BUTTON_1));
 #    ifdef MK_MOMENTARY_ACCEL
-    else if (code == QK_MOUSE_ACCELERATION_0)
+    else if (code == QK_MOUSE_ACCELERATION_0 && old_speed == mkspd_0)
         mk_speed = mkspd_DEFAULT;
-    else if (code == QK_MOUSE_ACCELERATION_1)
+    else if (code == QK_MOUSE_ACCELERATION_1 && old_speed == mkspd_1)
         mk_speed = mkspd_DEFAULT;
-    else if (code == QK_MOUSE_ACCELERATION_2)
+    else if (code == QK_MOUSE_ACCELERATION_2 && old_speed == mkspd_2)
         mk_speed = mkspd_DEFAULT;
     if (mk_speed != old_speed) adjust_speed();
 #    endif
```

This will prefer the most recently pressed acceleration profile, only reverting to unset if the button corresponding to the *current mouse speed* is released.

I'm not sure if this warrants a pull to the original repo or not. If so, I can make a request.