qmk/qmk_firmware #5699: QML Toolbox flashes XD75 successfully but keyboard will not work

df111, 2019-04-24:
I use QMK configurator ( https://config.qmk.fm/#/xd75/LAYOUT_ortho_5x15 ) to create my keymap for xd75re bought from kprepublic (the same model keyboard referenced in the qmk configurator code for the xd75). I configure the keymap to be simple as a test just assigning each a letter (nothing fancy no layers). I flash it to my xd75 using qmk toolbox. It is successful. See message below.

But when i type on my keyboard all sorts of windows pop open and weird stuff happens. Most keys do nothing.

I have spent hours on this. Going through every reddit post. I have limited knowledge so programming a c file myself and editing source code is out of the question. I basically just follow the noobs tutorial for the tools above and it all worked as it should. But the keyboard doesnt work.

*** DFU device connected: ATmega32U4 -- 03EB:2FF4:0001 {eb781aaf-9c70-4523-a5df-642a87eca567}

Something went wrong when trying to get the keyboard list from QMK.FM, you might not have a internet connection or the servers are down.

*** Attempting to flash, please don't remove device

>>> dfu-programmer.exe atmega32u4 erase --force

Erasing flash... Success

Checking memory from 0x0 to 0x6FFF... Empty.

>>> dfu-programmer.exe atmega32u4 flash "C:\Users\redacted\Documents\1-Personal\Filing Cabinet\Projects\Keyboard\XD75\xd75_layout_mine 2.hex"

Checking memory from 0x0 to 0x5EFF... Empty.

0% 100% Programming 0x5F00 bytes...

>>> dfu-programmer.exe atmega32u4 reset

[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>] Success

0% 100% Reading 0x7000 bytes...

[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>] Success

Validating... Success

0x5F00 bytes written into 0x7000 bytes memory (84.82%).

*** DFU device disconnected: ATmega32U4 -- 03EB:2FF4:0001 {eb781aaf-9c70-4523-a5df-642a87eca567}

drashna, 2019-04-25:
Just to make sure, does unplugging and replugging in the board fix the issue?

df111, 2019-04-28:
no it does not.

i have spent about 15 hours trying to make it work now. i have done hand stands in between pulling it out and plugging it back in. that doesnt work either

fauxpark, 2019-04-29:
Did you try resetting the EEPROM?

drashna, 2019-05-14:
Also, are you sure it's not this board? 

https://config.qmk.fm/#/idobo/LAYOUT_ortho_5x15

markstos, 2019-07-29:
There's been no further follow-up from the post author for a couple months. Should this be closed?

closed 2019-12-21 by stale[bot]