qmk/qmk_firmware #5700: MOD_MASK_* constants for single modifiers, and modifier macros in general

vomindoraan, 2019-04-24:
In #4337 we added the `MOD_MASK_CTRL`, `MOD_MASK_SHIFT`, `MOD_MASK_CSA` etc. constants to core code. These are shorthand for `(MOD_BIT(KC_LCTRL) | MOD_BIT(KC_RCTRL))` etc. since expressions of this form are very common.

Would constants for single modifiers, e.g. `#define MOD_MASK_LCTRL MOD_BIT(KC_LCTRL)`, be equally as useful? I think they could be. I still see a lot of checks that use `MOD_BIT()` with single modifiers in the codebase. Adding these definitions would “complete the set”, so to speak. It could also help reduce the confusion between:
- `MOD_MASK_*` constants (used with functions such as `get_mods()`, `set_mods()` etc.); and
- `MOD_*` constants (used with `MT()`, `LM()`, `OSM()` and internally in `tmk_core`).

The new constants could be implemented like this: https://github.com/qmk/qmk_firmware/compare/master...vomindoraan:mod_mask_single. Another thing to consider are the names. Should the new constants use:
- the long form, `MOD_MASK_LSHIFT` (consistent with `MOD_MASK_SHIFT`); or
- the short form, `MOD_MASK_LSFT` (consistent with `MOD_LSFT`, shorter and easier to align); or
- both, with the short names being aliases for the long names?

/discuss

vomindoraan, 2019-04-24:
@drashna @noroadsleft I'd appreciate your guys' input on this matter, since you seem to be using the `MOD_MASK_*` constants the most. @fauxpark too, since you were involved with the original PR.

drashna, 2019-04-25:
I think it would be useful, yeah.   I think it would be an edge case, though. But that's fine.

Also, I think that MEH and HYPER should be added, as well. :)

As for the naming, I think aliasing the various options would be best. That way either can be used, interchangeably.  Which is the current behavior for keycodes, and the like.

noroadsleft, 2019-04-25:
I can't speak to masks for MEH or HYPER, but I agree with everything else drashna said.

closed 2019-12-21 by stale[bot]