qmk/qmk_firmware #5701: Tap Dance key switching to strange layer

jasondunsmore, 2019-04-24:
I have my shift keys configured to send parenthesis upon single tap and brackets with double tap, and I'm having an issue where if I hold right shift and single tap left shift (I mistakenly did this when trying to type a "{"), the keyboard switches to a bizarre layer that isn't defined in my keymap.c file.

I mapped out the layer using keyboardtester.com (using pseudo-keycodes):

```
//,----+----+----+----+----+----.              ,----+----+----+----+----+----.
 NO,NO,CTRL_ALT,NO,CTRL_ALT,CTRL_ALT_D,         NO,  NO,  NO,  NO,  NO, SHIFT,
//|----+----+----+----+----+----|              |----+----+----+----+----+----|
 NO,CTRL_ALT,CTRL_ALT,NO,SHIFT_ALT,WIN,         SHIFT,NO,SHIFT, NO, SHIFT,NO,
//|----+----+----+----+----+----|              |----+----+----+----+----+----|
   VOLD, NO,  NO,  NO,  NO,  NO,                 NO,  NO,  NO,  NO, SHIFT,NO,
//|----+----+----+----+----+----+----.    ,----|----+----+----+----+----+----|
    NO,  NO,  NO,  NO,  NO,  NO, SHIFT,     NO,  NO,  UP,  NO,  NO,  NO,  NO,
//`----+----+----+--+-+----+----+----/    \----+----+----+----+----+----+----'
                      NO, SHIFT, NO,          NO,  NO,  UP 
//                  `----+----+----'        `----+----+----'
```

Here is my full keymap.c file:
https://github.com/jasondunsmore/qmk_firmware/blob/jasondunsmore/keyboards/keebio/iris/keymaps/jasondunsmore/keymap.c

I found that if I switch the Shift-Paren-Bracket keys to different physical keys on the keyboard, the issue still happens when I hold right shift and single tap left shift.

## System Information

 - Keyboard:  Keebio Iris
   - Revision (if applicable):  rev 2.8
 - Operating system: GNU/Linux (Ubuntu 18.04.2 LTS)
 - AVR GCC version: avr-gcc (GCC) 5.4.0

 - ARM GCC version: arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620

 - QMK Firmware version: 0.6.340 (latest master as of this morning)

 - Any keyboard related software installed? 
   - [ ] AutoKey (but already ruled it out)

jasondunsmore, 2019-04-24:
shinmai from Reddit did some testing on a Preonic and was unable to reproduce the issue:
https://www.reddit.com/r/olkb/comments/bgmrmh/qmk_tap_dance_freezing_up_keyboard/

noahfrederick, 2019-04-24:
See:
- https://docs.qmk.fm/#/feature_space_cadet_shift?id=caveats
- https://github.com/qmk/qmk_firmware/issues/4793

jasondunsmore, 2019-04-24:
Thanks for the pointer. I solved it with:
`COMMAND_ENABLE = no`

closed 2019-04-24 by jasondunsmore