qmk/qmk_firmware #5705: dz60rgb LED stays on while in sleep/suspend

Sada-Mio, 2019-04-25:
## Describe the Bug

RGB LED lights stays ON when computer goes into sleep/suspend mode. Tried on multiple desktop PC & Macbook

## System Information

 - Keyboard:  
   - Revision (if applicable):  v1
 - Operating system: Windows 10 & MacOS 10.14.3
 - AVR GCC version:  https://config.qmk.fm/#/dztech/dz60rgb/LAYOUT

 - ARM GCC version: https://config.qmk.fm/#/dztech/dz60rgb/LAYOUT

 - QMK Firmware version: https://config.qmk.fm/#/dztech/dz60rgb/LAYOUT

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

While in sleep mode the LED RGB animation effect is slower & less fluid.

drashna, 2019-04-26:
It's not probably sleeping, right now. 

However, it will take some code change to the dz60rgb to implement properly.

moyi4681, 2019-04-26:
@drashna thanks, already fixed.

drashna, 2019-05-14:
Awesome.

I'm closing this, as it appears to be resolved.  If not, please re-open.

closed 2019-05-14 by drashna