qmk/qmk_firmware #5722: SEND_STRING("#") sends '£' character.

johnfwhitmore, 2019-04-27:
Trying to create a keyboard macro key for C Programming language constructs and hit a snag with '#' character. This might be specific to a UK Keyboard layout, where Shift+3 is the '£' not '#' as in the US keyboard layout.

SEND_STRING("#include")

Results in '£include" 

I then tried to include unicode into the build and send with:
send_unicode_hex_string("0023 0069 006e 0063 006c 0075 0064 0065");
But that simply prints the numbers.

## System Information

 - Keyboard:  Satan 60
   - Revision (if applicable):  
 - Operating system: Ubuntu 16.04
 - AVR GCC version:  1:  avr-gcc (GCC) 4.9.2

 - ARM GCC version: 

 - QMK Firmware version:  0.6.340

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

fauxpark, 2019-04-27:
You probably want to `#include "sendstring_uk.h"` in your keymap.c.

johnfwhitmore, 2019-04-27:
I was just looking through code stumbled across that include file. Included it but makes no difference still getting '£'

fauxpark, 2019-04-27:
Ah, I see. £ = Shift+3 and # is actually Alt+3, but in sendstring_uk.h the ASCII code for # is marked in the Shift lookup table.

Try adding this to your keymap.c:

```c
const bool ascii_to_altgr_lut[128] PROGMEM = {
    0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0,

    0, 0, 0, 1, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0
};
```

pelrun, 2019-04-27:
Surely the altgr lut should go into sendstring_uk.h as well?

fauxpark, 2019-04-27:
Yeah, I just wanted to see if it actually worked because it sends KC_RALT.

johnfwhitmore, 2019-04-27:
Made those changes to 'ascii_to_altgr_lut' and 'ascii_to_shift_lut' which has stopped the '£' being output, but nothing has replaced it. So SEND_STRING("#include") outputs 'include'

pelrun, 2019-04-27:
Ok, I actually went and looked at the UK layout and it's not Alt+3 at all. So you can get rid of the altgr lut, and change the same entry in the ascii LUT from KC_3 to KC_NUHS. (At least, that's what keycodes_uk.h says it should be.)

fauxpark, 2019-04-27:
Derp, I was looking at some page talking about macOS, when I press Option+3 I get £. NUHS is the way to go.

johnfwhitmore, 2019-04-27:
YES! KC_NUHS thanks a million for your help.

diff --git a/quantum/keymap_extras/sendstring_uk.h b/quantum/keymap_extras/sendstring_uk.h
index 3c71f97..2824e5d 100644
--- a/quantum/keymap_extras/sendstring_uk.h
+++ b/quantum/keymap_extras/sendstring_uk.h
@@ -25,7 +25,7 @@ const bool ascii_to_shift_lut[0x80] PROGMEM = {
     0, 0, 0, 0, 0, 0, 0, 0,
     0, 0, 0, 0, 0, 0, 0, 0,
 
-    0, 1, 1, 1, 1, 1, 1, 0,
+    0, 1, 1, 0, 1, 1, 1, 0,
     1, 1, 1, 1, 0, 0, 0, 0,
     0, 0, 0, 0, 0, 0, 0, 0,
     0, 0, 1, 0, 1, 0, 1, 1,
@@ -45,7 +45,7 @@ const uint8_t ascii_to_keycode_lut[0x80] PROGMEM = {
     0, 0, 0, 0, 0, 0, 0, 0,
     0, 0, 0, KC_ESC, 0, 0, 0, 0,
 
-    KC_SPC, KC_1, KC_2, KC_3, KC_4, KC_5, KC_7, KC_QUOT,
+    KC_SPC, KC_1, KC_2, KC_NUHS, KC_4, KC_5, KC_7, KC_QUOT,
     KC_9, KC_0, KC_8, KC_EQL, KC_COMM, KC_MINS, KC_DOT, KC_SLSH,
     KC_0, KC_1, KC_2, KC_3, KC_4, KC_5, KC_6, KC_7,
     KC_8, KC_9, KC_SCLN, KC_SCLN, KC_COMM, KC_EQL, KC_DOT, KC_SLSH,

johnfwhitmore, 2019-04-27:
Since there's already a pull request in for this problem I assume this issue can be closed.

PR #5724 (2019-04-27): Fix sendstring "#" producing "£" instead

## Description

Using the # character in a `SEND_STRING` erroneously produces £ = Shift+`KC_3`. Both of these are wrong and it should simply be sending unshifted `KC_NUHS`.

Credit to @pelrun for figuring this out 👍 

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #5722

## Checklist

- [x] My code follows the code style of this project.
- [ ] My change r [... cut]
closed 2019-04-27 by drashna