qmk/qmk_firmware #5725: KC_MPLY	prioritizes youtube over native/default music player (spotify)

amesa0, 2019-04-27:
Simply put, KC_MPLY has, as of late been prioritizing youtube over native/default music player (spotify). 

As for how long it has been doing this, I can pinpoint it to Less than a month more than a week. 

I don't think it is inherently an issue with QMK but I believe the keycode has been targeting programs that should not be targeted. 

Ex case. I am playing spotify, then open up a youtube video. In order to listen to the video I try to pause spotify but the key targets the youtube video instead.  

I have spoken to at least two others that have noticed this issue.

Best.

drashna, 2019-04-27:
Unfortunately, I don't think this is a QMK Firmware issue. 

Specifically, the firmware sends the key to the OS, and the OS handles it.  So if you're seeing different behavior, it may be an update to the OS itself that has changed the behavior. 

You can test this, by rolling back to an older version of QMK Firmware, and testing to see if the issue persists.

amesa0, 2019-04-27:
Yes, I am just noticing that, I have spoken with an owner of a "cod67"? keyboard which does not run on QMK apparently and the same issue is occuring, It would seem to be OS or other issue that is not QMK firmware. 

I will leave the issue up instead of closing it in the case that a community member or qmk developer would like to design a workaround to target native players. :) (Or you can ping me to close the issue in the case you don't want clutter in the issues tab) 

Thank  you for the rapid reply drashna. 

Best.

drashna, 2019-04-27:
I suspected that much.  So it may be worth tracking down the update/change that caused that in your OS, and uninstall them.

bcat, 2019-04-29:
Chrome 73 [supports media keys](https://developers.google.com/web/updates/2019/02/chrome-73-media-updates#media-keys) now, at least on Windows/Mac/ChromeOS. So I think this is WAI. (I'm not sure if OSes have prioritization mechanisms for when multiple apps all want to respond to media keys, or if they just go to an arbitrary app.)

amesa0, 2019-05-01:
@bcat @drashna solved https://www.reddit.com/r/windows/comments/asdlwr/keyboard_media_controls_not_working_with/ 

Closing issue.

closed 2019-05-01 by amesa0