qmk/qmk_firmware #5735: NO_ACTION_TAPPING disables layers

jandk, 2019-04-29:
I was just playing around for fun, because I'm probably going to use taps at one point. But defining NO_ACTION_TAPPING disables the layer switching. Even with MO

drashna, 2019-04-30:
That's odd.

I can confirm that MO is disabled, but .... TG is not.  

Oh, okay.  MO is technically `ACTION_LAYER_TAP((layer), OP_ON_OFF)`.  So, we'd need to more finely ifdef it...

jandk, 2019-04-30:
Good to know, thought I went crazy for a minute 😊

drashna, 2019-04-30:
I mean, crazy is always a possibility, but this is definitely an unintended effect. 

I'll see about fixing this, if i can.

thinkv, 2019-05-08:
I was running into this issue as well, Clueboard 66 Hotswap prototype (atmega32u4), NO_ACTION_TAPPING was defined by default for this keyboard, so I had to undef it to enable my MO() layer tapping as well as LT() layer switching.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc

rileyjshaw, 2022-06-18:
@tzarc should this be closed yet? @drashna is still actively working on a PR: #11528.

tzarc, 2022-06-18:
> @tzarc should this be closed yet? @drashna is still actively working on a PR: #11528.

Was doing cleanup overall -- last activity on this issue was 3 years ago.

If we reopen it, it's going to get closed by stalebot after 120 days anyway -- if nobody has commented or updated this in 3 years I doubt another 4 months will be any different.

Besides, in my view there's no real need to keep this open if there's already a PR in progress. It's already being "tracked" in one way or another.

rileyjshaw, 2022-10-11:
Cool, makes sense to me!

arretxe, 2022-11-10:
Not sure how much code size NO_ACTION_TAPPING saves, but I need Layers and not ACTION_TAPPING so I would really be itnerested in having this corrected, maybe I can try myself looking where the NO_ACTIONG_TAPPING ifdefs are

PR #11528 (2023-02-19): Fix functions when `NO_ACTION_TAPPING` is defined
## Description

Changes so that when `NO_ACTION_TAPPING` is defined, that layer function act as expected.  Eg, that MO and TG still work. 

* LT's work as just their keycode
* MT's work as their keycode (but can be switched to mods)
* One Shot Layers work as MO
* One Shot Mdds work as their mod
* TT works as TG (but can be switched to MO
* Tap Dance works (with a fix), considering disabling outright as it is heavily "tapping" dependant
* File formatted

## Types of Changes
- [x] Core
- [x] Bugfix
- [x] Enhancement/optimization
- [ ] Documentati [... cut]