qmk/qmk_firmware #5761: How should LED_FLAG_MODIFIER be used with RGB Matrix?

jpetermans, 2019-05-02:
There is some great changes happening with the RGB Matrix code which is very exciting. This is just a general question on the intent of the LED_FLAG_MODIFIER flag. Is that to designate modifier keys as in Shift, Ctrl, Alt, etc, separate from Alpha keys?

I'm using the massdrop ALT and the only flags used under config_led.c are KEYLIGHT and UNDERGLOW.  Would using MODIFIER in the map make it more complicated to turn on leds for mods and alphas together?

noroadsleft, 2019-05-02:
I believe @XScorpion2 is best equipped to answer this.

XScorpion2, 2019-05-02:
ALT has 1 key set up as a MODIFIER already, so it doesn't make it more complicated to turn on leds for mods and alphas together as it already does so.

Think of it this way, in rgb matrix there is a set and get flags API, this tells the rgb matrix code which LED types to use for the running effect, so if you set KEYLIGHT  | MODIFIER, it will run the effect on both mods and keys, but if you set just KEYLIGHT, it will run on just keys, no mods

jpetermans, 2019-05-02:
Thanks for the response. I missed the one "1" in the led table.  I understand you can flag whatever keys you want, but is the intent of that flag any regular modifiers or something more specific like Caps/Num Lock? 

Using the bit flags KEYLIGHT | MODIFIER makes total sense for combining effects.

What is the best way to customize these flags for your own keymap?  Is config_led.c overridden if tweaked in a personal keymap folder?

XScorpion2, 2019-05-02:
The intent really is left up to keyboard designers on what they want to flag as what. I have an Ergodox like keyboard that I worked on the firmware for and setup thumb cluster and all the keys in the U shape as modifiers. Probably overkill, but it looks good.

As for the best method to customize these flags, I've been on the fence tbh. I removed the const flag on the deceleration so you can manipulate the data at any point during runtime so you could do so in init or during keymap layer change as an example. I've thought of adding the weak attribute to them all, but I feel like that is just going to make things really messy and hard to keep user keymaps updated with their keyboards if the values got changed for one reason or another.

XScorpion2, 2019-05-02:
As far as current flags, right now we only have 1, 2, and 4 defined, so that leaves 5 more flags you could use for any crazy thing you want to do (8, 16, 32, 64, 128). I would suggest for custom user flags starting at the highest flag and work backwards in case we need to add more to qmk itself.

jpetermans, 2019-05-02:
No easy answers on customizing at this point it seems, but thanks for the feedback. Alternatively, it shouldn't be too bad to update groups of keys with a predefined array in individual keymaps. For now I just have a variable that mimics the existing qmk flags and adds a few more like you suggested on the other bits.

closed 2019-05-02 by jpetermans