qmk/qmk_firmware #5762: Multiple random keypresses after KC_LEAD freezes keyboard.

jpetermans, 2019-05-02:
I'm not sure I'd characterize this as a bug or just misuse of the LEADER feature. If I hit KC_LEAD then mash several keys quickly my keyboard completely freezes. I realize this isn't the intent of the LEADER feature, but it's not an implausible scenario.

I have a layer that locks my keyboard so anyone (like say a 2 year old) can mash on the keyboard without registering keypresses. It's unlocked with KC_LEAD ...  I've found an easy way to avoid the freeze issue, but didn't know if this is an expected behavior that just should be avoided.

## System Information
Massdrop ALT on Windows
QMK release 0.6.341

jpetermans, 2019-05-03:
Maybe an overflow of `leader_sequence[leader_sequence_size]` in `process_leader.c` if too many keypresses register before LEADER_TIMEOUT? I have very little c++ to know a protection for this but that seems like a plausible reason for the crash.

shinmai, 2019-05-10:
Have this same issue. From testing, it seems around 11-12 keypresses will freeze the board. Tested a dozen or so times by just hitting KC_LEAD followed by hitting one key on the board an increasing number of times.
Tested on a rev 6.1 Planck and a rev 3 Preonic.

shinmai, 2019-05-10:
Well, I seem to be posting horrible hacky workarounds to QMK issues all over the place tonight.
Adding
```
if(leader_sequence_size > 5) {
  leader_time = timer_read() - LEADER_TIMEOUT;
  return false;
}
```
after [`leader_sequence_size++;`](https://github.com/qmk/qmk_firmware/blob/bb1b44132514fc30de7de77f2c6a80b8c7ba0416/quantum/process_keycode/process_leader.c#L62) in `qmk_firmware/quantum/process_keycode/process_leader.c` will fail the leader after 6 keypresses (by running out the timer, because that was the fastest way I could think up to do it without venturing into how exactly the leader stuff is implemented :D), and keeps the board from locking up.

drashna, 2019-05-10:
Yup, that would be the best way, since that's what the actual documentation uses. 

Also, there's the problem:

```c
uint16_t leader_sequence[5] = {0, 0, 0, 0, 0};
```

That's already at 5, so it causes an overflow, I think.

drashna, 2019-05-10:
Actually, if you could, try changing this line:
https://github.com/qmk/qmk_firmware/blob/master/quantum/process_keycode/process_leader.c#L61

```c
        if (leader_sequence_size < 5) {
          leader_sequence[leader_sequence_size] = keycode;
        } else {
          leading = false;
          leader_end();
        }
```

jpetermans, 2019-05-10:
Makes sense to me.

drashna, 2019-05-10:
well, it's adding a keycode to index 6, on an array that only has 5 entries.  So ... yeah.  

I don't use leader keys, but if you could test this fix, (and just it), and let me know.

PR #5840 (2019-07-16): Add out of bound check for Leader Key sequence array
Leader keys is limited to 5 keypresses, even the array for is sets it to 5, but at no time does it check to make sure that it's hit the end of the array before adding more key sequences.... which appears to crash the controller. 

## Description

This adds a sanity check to make sure that once you've hit 5 keys, it doesn't add more.  However, it forcibly ends the lead sequence, and may not process it correctly. 

## Types of Changes

- [x] Core
- [x] Bugfix

## Issues Fixed or Closed by This PR

* Fixes #5762

## Checklist

- [x] My code fol [... cut]
closed 2019-07-16 by skullydazed