qmk/qmk_firmware #5776: [WIP] Python support in the build system

skullydazed, 2019-05-04:
This is a work in progress. Everything below is currently subject to change. Please comment if you have any concerns or questions about the plan below.

# Python as a build dependency

We've been talking about some of the deficiencies of make and shell for a long time. We've talked a lot about what language would be a suitable choice to make certain tasks easier, and from those talks python has emerged as the clear winner.

This issue exists to collect that conversation into one place so we can make a decision and move forward. 

## Background

We have discoverability and build env issues. Adding new functionality using make and shell is growing extremely complex. Pulling the user-facing features out of our make system will simplify it a lot and make it easier to add new functionality.

There are a lot of languages/frameworks that could be used for this, but right now the preference seems to trend towards python. It's a widely available language that's already installed on the vast majority of client systems. On the systems where it's not already installed it could be installed with a single command.

Given our list of platforms we should target python 3.5. This is the oldest version currently shipped by every platform we support.

3rd Party Modules: Whenever possible we should use built-in modules. There are a few exceptions for functionality that is not included in python:

* jinja2 for templates

# Action Items

* [ ] Add python to all OS dependencies
* [ ] Identify functionality to rewrite in python
  * [ ] new_project.sh
  * [ ] new_keymap.sh

ymzcdg, 2019-05-24:
em,I don't know what this issue is. But I prefer scons. Lots of Chinese programmer use IDE like keil, IAR etc. Scons can create the project file of these IDE.
Although I can neither master makefile nor scons(maybe I am reckless, haha).

yanfali, 2019-05-31:
Can we close this yet?

drashna, 2019-05-31:
yup

closed 2019-05-31 by drashna